greatscottgadgets/ubertooth #320: Error wireshark plugins btbb

shabrankhairi, 2018-07-05:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
i just follow step in build guide

### Actual behaviour
error in make command after cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins .. command

### Version information
**Operating system**: Ubuntu 18.04

**Ubertooth tools version (ubertooth-rx -V): libubertooth 1.0 (2018-06-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v): Firmware version: 2018-06-R1 (API:1.03)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output

/libbtbb-2018-06-R1/wireshark/plugins/btbb/build$ make
[ 20%] Generating plugin.c
Scanning dependencies of target btbb
[ 40%] Building C object CMakeFiles/btbb.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
In file included from /usr/include/wireshark/epan/wmem/wmem_array.h:18:0,
                 from /usr/include/wireshark/epan/wmem/wmem.h:15,
                 from /usr/include/wireshark/epan/proto.h:30,
                 from /usr/include/wireshark/epan/packet.h:14,
                 from /home/vault/libbtbb-2018-06-R1/wireshark/plugins/btbb/packet-btbb.c:32:
/usr/include/wireshark/epan/wmem/wmem_core.h:18:10: fatal error: ws_attributes.h: No such file or directory
 #include "ws_attributes.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/btbb.dir/build.make:93: recipe for target 'CMakeFiles/btbb.dir/packet-btbb.c.o' failed
make[2]: *** [CMakeFiles/btbb.dir/packet-btbb.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/btbb.dir/all' failed
make[1]: *** [CMakeFiles/btbb.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Insert any commandline or build output here
```

dominicgs, 2018-07-28:
Which version of Wireshark?
Can we see the output from the `cmake` command?

closed 2020-12-05 by mikeryan

Zarathustraa, 2023-04-16:
same prob here missing config.h in /usr/include/wireshark/......