greatscottgadgets/ubertooth #321: Ubertooth one not working

Grassyjansen, 2018-07-05:
### Steps to reproduce
1. After an update of the ubertooth one the device worked
2. Now a few months the device is not showing on the usb bus when inserted
3. DFU mode by shorting the 2 pins also does not work
4. Tried to flash via the EXPAND port. Connected an FTDI to the RX, TX and GND and shorted CTS to GND
5. Then tried lpc21isp -detectonly /dev/ttyUSB0 115200 4000000 and 
lpc21isp -detectonly /dev/ttyUSB0 115200 14746

but "nswer on '?'

What am I doing wrong?

mikeryan, 2018-07-05:
Do any LEDs turn on when you plug it into USB? What color(s)?

dominicgs, 2018-07-05:
In addition to Mike's question about:
I'm rather surprised that DFU mode doesn't work.  What happened to the Ubertooth between steps 1 and 2?

Is there any sign of damage to the USB connector?

Grassyjansen, 2018-07-09:
2 green lights turn on green after plugging it in:
1.8V
USB led

There is no sign of damage on the USB connector

dominicgs, 2018-07-09:
Does anything show up in the output of dmesg when you connect the Ubertooth?

Grassyjansen, 2018-07-13:
No nothing

Grassyjansen, 2018-07-18:
Does this mean the ubertooth is beyond repair? Is there a possibility to flash via the expand port?

dominicgs, 2018-07-28:
I'm rather baffled by this, the bootloader is there for just this sort of situation and it was clearly working initially because you were able to update the firmware (which requires the bootloader).

Where di you buy your Ubertooth from?

Grassyjansen, 2018-09-07:
Good question, it was bought by a colleague who does not work at the company any more. Is there another way to flash the device? For example by flashing via the EXPAND port?

mikeryan, 2020-12-05:
@Grassyjansen if you're still having problems with your Ubertooth, please reopen this issue and we can help you debug it further.

closed 2020-12-05 by mikeryan