greatscottgadgets/ubertooth #322: Ubertooth Scan fails.

ramprasad1211, 2018-07-09:
### Steps to reproduce
1. I followed build guide for installing ubertooth
2. I run the command "ubertooth-scan -x" or "ubertooth-scan -s"
3.

### Expected behaviour
Tell us what you expect should happen : I expect the bluetooth devices to be listed. 

### Actual behaviour
Tell us what happens instead: I get error: 
root@banwifi-146686:~# ubertooth-scan -s
error: Unable to find hci0 (-1)
root@banwifi-146686:~# ubertooth-util -N
1

### Version information
**Operating system**: Ubuntu

**Ubertooth tools version (ubertooth-rx -V):**

libubertooth 1.0 (2018-06-R1), libbtbb 1.0 (git-d615b97)

**libbtbb version:**

libubertooth 1.0 (2018-06-R1), libbtbb 1.0 (git-d615b97)

**Ubertooth firmware version (ubertooth-util -v):**

Firmware version: 2018-06-R1 (API:1.03)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
root@banwifi-146686:~# ubertooth-scan -s
error: Unable to find hci0 (-1)
root@banwifi-146686:~#

Where as other commands like ubertooth-specan or ubertooth-btle works fine. 
```

ramprasad1211, 2018-07-09:
Even hcitool dev doesn't list this device.

hcitool dev
Devices:

mikeryan, 2018-07-09:
Ubertooth is not a Bluetooth dongle and as such won't show up in hcitool or hciconfig. We've documented that in our primary man page: https://github.com/greatscottgadgets/ubertooth/blob/master/host/doc/ubertooth.md

`ubertooth-scan` works in conjunction with a regular Bluetooth dongle to display information about discovered Bluetooth devices. For more info on how it works, refer to its manpage: https://github.com/greatscottgadgets/ubertooth/blob/master/host/doc/ubertooth-scan.md

If you want to simply scan for non-discoverable devices without using a Bluetooth dongle, use `ubertooth-rx -z -t 20`.

closed 2020-12-05 by mikeryan