greatscottgadgets/ubertooth #323: BR Packets parsing

nitindhingramobiliya, 2018-07-11:
I am running command "ubertooth-rx -l aba2b2 -q capture_pcap.log -r capture_pcap_ng.log", I get a dump with no issues. I open up on Wireshark with one of these files, it opens showing the packet information. 

But this information is not good enough for me, as it is not showing Bluetooth profile/protocol level information (LMP, A2DP, OPP etc.)

My main purpose is only to check LMP command requests packets, how is it possible?

Please respond soon. 

Thanks

dominicgs, 2018-07-11:
That capture is almost certainly encrypted and quite possibly incomplete, so we are unable to extract profile and protocol level information from the packets.

nitindhingramobiliya, 2018-07-11:
Thanks for the reply.
Actually, I am just looking for the response from the slave for the LMP encryption key size req. The encryption will start after this.

nitindhingramobiliya, 2018-07-13:
I tried many times, but I am not getting packets (BR) that are exchanged before connection completion. Is there a limitation in ubertooth for this?

dominicgs, 2018-07-13:
The packets before the connection is established won't necessarily have the LAP that you're looking for.  The hopping pattern is different before the connection has been established too, so we can't follow it.

Yes, this is a limitation of Ubertooth (as well as every <$20,000 Bluetooth sniffer that I've seen)

drobinson22222, 2018-09-06:
Dominic said - "That capture is almost certainly encrypted and quite possibly incomplete, so we are unable to extract profile and protocol level information from the packets."

closed 2020-12-05 by mikeryan