greatscottgadgets/ubertooth #324: ubertooth-btle-libUSB-Error-Timeout-(-7)

ChenYenTing, 2018-07-16:
Hello

When I input " sudo ubertooth-btle -f " in virtual machine , it can run for a while and show this error.
How can I solve it ?
[image]

pckbls, 2018-09-11:
See: Issue https://github.com/greatscottgadgets/ubertooth/issues/174

Best regards,
Patrick

pckbls, 2018-09-11:
I cannot find the exact issue number but the developer once stated that there are a number of problems when trying to use the Ubertooth One inside a virtualized environment. He suggested to use a native Linux machine instead.

mikeryan, 2019-01-13:
Indeed, Ubertooth runs poorly inside a VM and this type of timeout is one of the known issues. If you have a native OS X or Linux system, use that.

closed 2019-01-13 by mikeryan