greatscottgadgets/ubertooth #326: Ubertooth One libUSB error (-9): USB error

celosiag, 2018-07-30:
### Steps to reproduce
1. Flashing the Ubertooth One with the 2018-06-R1 firmware
2. sudo ubertooth-btle -f

### Expected behaviour
Scan for nearby BLE advertisements until a CONNECT_REQ is received (after the reception of the CONNECT_REQ, the Ubertooth will follow the connection between the two devices)

### Actual behaviour
Crash during the BLE advertisements scanning although any CONNECT_REQ was received

### Version information
**Operating system**: Linux 4.15.0-24-generic #26~16.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux

**Ubertooth tools version (ubertooth-rx -V)**: libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)

**libbtbb version**: 2018-06-R1

**Ubertooth firmware version (ubertooth-util -v)**: Firmware revision: 2018-06-R1

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
systime=1532942998 freq=2402 addr=8e89bed6 delta_t=11.236 ms
42 25 2c 8d 18 a5 97 20 62 79 86 36 4e 94 d7 b1 08 ec 32 d9 fb 10 29 4d d8 14 3c 29 f0 58 38 bf 6d aa 2c b5 88 e2 99 90 7b 48 
Advertising / AA 8e89bed6 (valid)/ 37 bytes
    Channel Index: 37
    Type:  ADV_NONCONN_IND

    Data:  2c 8d 18 a5 97 20 62 79 86 36 4e 94 d7 b1 08 ec 32 d9 fb 10 29 4d d8 14 3c 29 f0 58 38 bf 6d aa 2c b5 88 e2 99
    CRC:   90 7b 48

libUSB Error: Command Error:  (-9)
USB error

```

celosiag, 2018-07-30:
**New output after compiling and installing the host Ubertooth tools from the 2018-06-R1 release package** :
```
systime=1532944003 freq=2402 addr=8e89bed6 delta_t=2.086 ms rssi=-92
42 25 e1 c7 79 a4 42 1b 62 79 86 36 4e 94 8c 96 60 0b a2 5f bc e2 38 62 88 eb 51 ba 34 ae 3a 6b 23 94 5a 5d b9 5e 15 de 42 40 
Advertising / AA 8e89bed6 (valid)/ 37 bytes
    Channel Index: 37
    Type:  ADV_NONCONN_IND
    AdvA:  1b:42:a4:79:c7:e1 (random)
    AdvData: 62 79 86 36 4e 94 8c 96 60 0b a2 5f bc e2 38 62 88 eb 51 ba 34 ae 3a 6b 23 94 5a 5d b9 5e 15
Error: attempt to read past end of buffer (1 + 98 > 31)

    Data:  e1 c7 79 a4 42 1b 62 79 86 36 4e 94 8c 96 60 0b a2 5f bc e2 38 62 88 eb 51 ba 34 ae 3a 6b 23 94 5a 5d b9 5e 15
    CRC:   de 42 40

libUSB Error: Pipe error:  (-9)
USB error

```

mikeryan, 2018-07-30:
Try the latest master firmware and host tools, this bug has been fixed
there.

celosiag, 2018-07-31:
It's what I do and now the error is different. Instead of having the "libUSB Error: Pipe error:  (-9)", I have no error but the output of ubertooth-btle seems to freeze after a session (~30-40mins) of btle scanning.

### New version information
**Operating system**: Linux 4.15.0-24-generic #26~16.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux

**Ubertooth tools version (ubertooth-rx -V)**: libubertooth 1.1 (git-54109a7), libbtbb 1.0 (git-d615b97)

**libbtbb version**: 1.0

**Ubertooth firmware version (ubertooth-util -v)**: Firmware version: git-54109a7* (API:1.04)

### Output
```
systime=1533030880 freq=2402 addr=8e89bed6 delta_t=0.397 ms rssi=-34
00 24 11 00 11 00 11 00 02 01 06 1a ff 4c 00 02 15 b9 40 7f 30 f5 f8 46 6e af f9 25 55 6b 57 fe 6d 00 01 00 02 b6 f5 8f e3 
Advertising / AA 8e89bed6 (valid)/ 36 bytes
    Channel Index: 37
    Type:  ADV_IND
    AdvA:  65:d1:10:41:3e:fb (random)
    AdvData: 02 01 06 1a ff 4c 00 02 15 b9 40 7f 30 f5 f8 46 6e af f9 25 55 6b 57 fe 6d 00 01 00 02 b6
        Type 01 (Flags)
           00000110
               LE General Discoverable Mode
               BR/EDR Not Supported

        Type ff (Manufacturer Specific Data)
           Company: Apple, Inc.
           Data: 02 15 b9 40 7f 30 f5 f8 46 6e af f9 25 55 6b 57 fe 6d 00 01 00 02 b6

    Data:  11 00 11 00 11 00 02 01 06 1a ff 4c 00 02 15 b9 40 7f 30 f5 f8 46 6e af f9 25 55 6b 57 fe 6d 00 01 00 02 b6
    CRC:   f5 8f e3

```

ChenYenTing, 2018-09-06:
Excuse me do you know what does this "Error : attempt to read past end of buffer" mean ?
Is this error a big trouble ?

davidHisense, 2018-11-01:
The same problem for me,whether the bug has been fixed or not?
Firmware version: 2018-08-R1 (API:1.05)

celosiag, 2018-11-01:
@ChenYenTing : 
(1) I don't know what this error means but you can find it in this libbtbb [C file](https://github.com/greatscottgadgets/libbtbb/blob/master/lib/src/bluetooth_le_packet.c) (at line 408). In the past, I had the same error but flashing the Ubertooth One with the already compiled 2018-08-R1 [firmware](https://github.com/greatscottgadgets/ubertooth/releases/tag/2018-08-R1) (in _ubertooth-one-firmware-bin/_ directory) solved it. 
(2) To sniff BLE data with _ubertooth-btle_, you have to use this tool with the -f (follow) argument in order to firstly capture a CONNECT_REQ between two devices. After capturing this CONNECT_REQ, the Ubertooth will follow the connection between the two devices and sniff the exchanged data. Otherwise, as you noticed, it only sniff "broadcast packages" (called "BLE advertisements").

@davidHisense : To me, the "bug" is always here. I've tried to code from scratch different related _ubertooth-btle_ tools (with a selfmade FIFO queue for BLE packet management, for instance) but I never managed to fix this bug... If you have any tracks on this bug's origin, I'm interested to test other implementations and give you feedback :).

nick990, 2018-11-21:
Same bug for me.
Firmware version: 2018-08-R1 (API:1.05)
Running `ubertooh-btle -n` to scan BLE advertisements, it seems to freeze (with no error message) after a session of variable length.
During my last tests I've obtained several sessions duration before freezing: 6 min, 15 min, 28 min, 1.10h.

nick990, 2018-11-27:
@mikeryan 
> Try the latest master firmware and host tools, this bug has been fixed there.

I've followed the [build guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide) which 
refers to the latest release 2018-08-R1 and the bug is still present. Is there a new release?

confilemg, 2018-12-01:
I have the same problem. Firmware version: 2018-08-R1 (API:1.05) but I get only broadcast packages.

mostafau, 2019-04-15:
It seems after certain time/or for certain cause ubertooth firmware packet queue gets empty. Therefore ubertooth-btle doesn't get any packet from the ubertooth one. I was wondering can there be a issue of handling unexpected size packet in  DMA handler at 'le_DMA_IRQHandler' ??

closed 2019-11-08 by mikeryan

sogewasp, 2020-01-21:
Same problem here with 2018-12-R1. After less than 5min capture the output freezes and I get nothing else and I have to stop and restart.

Ubertooth tools were installed from Kali Linux packages:
```
$ apt show ubertooth
Package: ubertooth
Version: 2018.12.R1-3
Priority: optional
Section: science
Maintainer: Ruben Undheim <ruben.undheim@gmail.com>
Installed-Size: 311 kB
Depends: libubertooth1 (= 2018.12.R1-3), libbluetooth3 (>= 4.91), libbtbb1 (>= 2015.10.R1+20161027git1eecca5), libc6 (>= 2.7), libusb-1.0-0 (>= 2:1.0.8)
Recommends: ubertooth-firmware
Suggests: ubertooth-firmware-source
Homepage: http://ubertooth.sourceforge.net/
Download-Size: 66.3 
...
```

```
$ uname -a
Linux kali 5.3.0-kali2-amd64 #1 SMP Debian 5.3.9-3kali1 (2019-11-20) x86_64 GNU/Linux
```

attila123, 2022-09-16:
Hello, I had the same issue all the time in an environment where I measured about 750 BTLE packets / second (on channel 37 alone, did not check the others). The capturing stopped in minutes. Sometimes it could run for several minutes, but I did not try to see the max time etc as I was busy.
I had the Ubertooth One firmware upgraded to the latest one. The ubertooth package is installed from AUR (https://aur.archlinux.org/packages/ubertooth) pkgbuild script: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ubertooth (latest change from 2021-01-13), and it's 2020.12.R1-1.

commit fea9d0fc (2019-11-08): le: disable DMA interrupt to workaround timer interrupt issues

Under certain circumstances, it is possible for a timer interrupt to
fire while a DMA interrupt is being handled. Regardless of interrupt
priority and subpriority between the DMA and timer interrupt, the timer
interrupt will never be serviced.

The LE sniffing code uses the timer interrupt to poll the CC2400's
FS_LOCK status. If this bug is triggered, the FS_LOCK polling will not
occur and the timer interrupt will not be rescheduled, causing capture
to freeze.

Curiously this only occurs after all DMA channels have been disabled,
 [... cut]