greatscottgadgets/ubertooth #327: Ubertooth-btle no longer follows null targets

anotherfiz, 2018-08-02:
### Steps to reproduce
1. ubertooth-btle -t00:00:00:00:00:00
2. ubertooth-btle -A37 -f

### Expected behaviour
Ubertooth should continuously sniff advertising data on channel 37 and not attempt to follow any connection requests

### Actual behaviour
Ubertooth does not sniff any packets

### Version information
**Operating system**: Ubuntu 18.0.4 LTS

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.0 (2018-06-R1)

**libbtbb version:**
libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**

### Output
```ubertooth-btle -t00:00:00:00:00:00
target set to: 00:00:00:00:00:00
ubertooth-btle -A37 -f
<nothing>

mikeryan, 2018-08-02:
This is a known issue. I plan to add a new command line option for advertising-only, perhaps -n for no-follow. Thoughts?

anotherfiz, 2018-08-02:
Sounds great to me!

commit fec3f3dd (2018-08-04): le: introduce feature to NOT follow connections

This oft-requested feature uses the LE sniffing engine to log
advertising packets but NOT follow connections as they are observed.
Previously we recommended people use a hack involving filters, but the
new LE sniffing engine made that hack non-functional. Instead we now
formally support explicitly requesting not to follow connections.

Closes #327
closed 2018-08-04 by mikeryan