greatscottgadgets/ubertooth #330: Freeze when there too many advs going on

sajingeo, 2018-08-21:
### Steps to reproduce
1. have 100 BLE beacons advertising at 100 ms, Interval
2. try to run ubertooth-btle -f -c /tmp/pipe
3.

### Expected behaviour
You should the BLE ADV_IND packets on the screen. When I have less BLE traffic everything works fine!!

### Actual behaviour
It just sits there!! with zero output.

### Version information
MAC OSx and Kali

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (2018-08-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:**
libbtbb-2018-06-R1

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2018-08-R1 (API:1.05)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
ubertooth-btle -f -c /tmp/pipe 

```

[image]

mikeryan, 2018-08-21:
In your screenshot, the Wireshark window does not show that it is capturing. Can you confirm that you have clicked the "start" button?

sajingeo, 2018-08-21:
lol my bad. Very stupid of me. Good catch!!

closed 2018-08-21 by sajingeo