greatscottgadgets/ubertooth #335: Build install failure on Ubuntu 16.04 (vm) Release 2018-08-R1 with WS plugin

woodm528, 2018-10-16:
Appears to be invalid header dependency building wireshark plugins

Running build procedure at https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide
Release 2018-08-R1: the DEFCON release

Ubuntu 16.04 	(vm on windows 10 – vmware workstation 12.5.7 build 5813279)
uname -r
4.4.0-137-generic

Wireshark Version 2.2.6:
Compiled (64-bit) with Qt 5.5.1, with libpcap, with POSIX capabilities (Linux),
with libnl 3, with GLib 2.48.0, with zlib 1.2.8, with SMI 0.4.8, with c-ares
1.10.0, with Lua 5.2.4, with GnuTLS 3.4.10, with Gcrypt 1.6.5, with MIT
Kerberos, with GeoIP, with nghttp2 1.7.1, with QtMultim

Commands & Trace:
#Wireshark ver 1.12 and newer includes ubertooth BLE plugin by default
sudo apt-get install wireshark wireshark-dev libwireshark-dev cmake
#aborted -space issue – to correct, ran:
	sudo apt autoremove
cd libbtbb-2018-08-R1/wireshark/plugins/btbb
mkdir build
cd build
cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins ..
#Also tried; (wireshark plugin path for this environment) same result
#cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/plugins .. 
make
[ 20%] Generating plugin.c
Scanning dependencies of target btbb
[ 40%] Building C object CMakeFiles/btbb.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
In file included from /usr/include/wireshark/epan/wmem/wmem_array.h:18:0,
                 from /usr/include/wireshark/epan/wmem/wmem.h:15,
                 from /usr/include/wireshark/epan/proto.h:30,
                 from /usr/include/wireshark/epan/packet.h:14,
                 from /home/mwood/ubertooth/libbtbb-2018-08-R1/wireshark/plugins/btbb/packet-btbb.c:32:
/usr/include/wireshark/epan/wmem/wmem_core.h:18:27: fatal error: ws_attributes.h: No such file or directory
compilation terminated.
CMakeFiles/btbb.dir/build.make:93: recipe for target 'CMakeFiles/btbb.dir/packet-btbb.c.o' failed
make[2]: *** [CMakeFiles/btbb.dir/packet-btbb.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/btbb.dir/all' failed
make[1]: *** [CMakeFiles/btbb.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

### Steps to reproduce
Follow build procedure at https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

### Expected behaviour
make command should complete without failure

### Actual behaviour
Make failure refer to trace above

### Version information
**Operating system**:
Ubuntu 16.04 	(vm on windows 10 – vmware workstation 12.5.7 build 5813279)
uname -r
4.4.0-137-generic

Wireshark Version 2.2.6:
Compiled (64-bit) with Qt 5.5.1, with libpcap, with POSIX capabilities (Linux),
with libnl 3, with GLib 2.48.0, with zlib 1.2.8, with SMI 0.4.8, with c-ares
1.10.0, with Lua 5.2.4, with GnuTLS 3.4.10, with Gcrypt 1.6.5, with MIT
Kerberos, with GeoIP, with nghttp2 1.7.1, with QtMultim

**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
See command / trace above

Thanks
Mike

davidHisense, 2018-10-17:
Hi,I also use the  same environment to install Ubertooth and Wirshark plugin,and it runs normally.

Firstly,please confirm the header file "ws_attributes.h" ? My path is: /usr/include/wireshark/ws_attributes.h

From your steps, the different of mine is the operation 'autoremove'.

Could you try reinstall wireshark, and then do not use the 'autoremove'  command to uninstall packages.This would delete some dependent packages.

BR.

mikeryan, 2019-01-13:
Closing this due to inactivity. If you continue to have trouble building this plugin please reopen.

closed 2019-01-13 by mikeryan