greatscottgadgets/ubertooth #337: Can not capture btsmp data package

davidHisense, 2018-11-01:
### Steps to reproduce
1.Running the ubertooth-btle -f -r pairing.pcapng
2.Open the bluetooth and try to connect my MacBook and Iphone 6.
3.When Display the Confirm options with PIN code, select "Confirm Pair".
4.Capturing all the time during last pair and connect step.And after 10 minutes,stop capturing.

Other device pairing and connecting test:
Iphone 6 vs  Android Phone(Konka 801)
Mac Book vs Android Phone(KonKa 801)
Iphone 6 vs Vehicle-BT device
Android Phone(Konka 801) vs Vehicle-BT device

### Expected behaviour
1.Open the "pairing.pcapng"  file with Wireshark tool.
2. And then input the filter "btsmp" to find the Bluetooth Secure Manager Protocol package.
3.It can display Pairing Request,Pairing Response,Pairing Random packages.

### Actual behaviour
When i input the filter "btsmp", i could not find any packages, while many advertise packages captured.

### Version information
Operating system:
Ubuntu 16.04

Ubertooth tools version (ubertooth-rx -V):
libubertooth 1.1 (2018-08-R1), libbtbb 1.0 (2018-06-R1)

libbtbb version:
libubertooth 1.1 (2018-08-R1), libbtbb 1.0 (2018-06-R1)

Ubertooth firmware version (ubertooth-util -v):
Firmware version: 2018-08-R1 (API:1.05)

###Need Help
Is there something wrong with my operation? Thanks.

mikeryan, 2019-01-13:
This means you probably didn't capture a connection. Try the Wireshark filter "btle.data_header". If there are zero packets, you did not capture the connection. You may need to repeat connecting and pairing several times, and even if you capture the connection it's possible you will still miss some packets. See the [Crackle FAQ](https://github.com/mikeryan/crackle/blob/master/FAQ.md#crackle-is-complaining-about-missing-packets-why-cant-i-crack) if your plan is to crack the pairing.

closed 2019-01-13 by mikeryan

saurabhs0294, 2019-08-08:
Hii, can you please share me Stock ROM of Konka with NVRAM data and bin my Konka 801 phone IMEI number is showing null and I don't have a backup.