greatscottgadgets/ubertooth #340: no ubertooth-one-firmware-bin in new release

bosworthc1019, 2018-12-07:
### Steps to reproduce
1. I have installed the new 2018-12-R1 release, followed instructions and no directory called ubertooth-one-firmware-bin
2. 
3.

### Expected behaviour
Tell us what you expect should happen
I expect it to be there
### Actual behaviour
Tell us what happens instead
All there is is a firmware directory, no firmware-bin
### Version information
**Operating system**:
linux mint terra
**Ubertooth tools version (ubertooth-rx -V):**
ubertooth 2017-03-R2 
**libbtbb version:**
libbtbb-2018-12-R1
**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2018-12-07:
Which archive are you downloading?  There's an ubertooth-one-firmware-bin directory in the archive here: https://github.com/greatscottgadgets/ubertooth/releases/download/2018-12-R1/ubertooth-2018-12-R1.tar.xz

bosworthc1019, 2018-12-08:
Awesome got it, thx!