greatscottgadgets/ubertooth #345: Ubertooth-btle understanding displayed frequency value

nishant412, 2019-01-09:
I am trying to use ubertooth-btle application with Ubertooth One in channel following mode for a defined access address. My attempt is to be able to easily read the frequency offset from the CC2400 whenever it receives a new data packet and display over the console along with frequency value. From my understanding of the code base, I can do this inside cb_follow_le by reading the FREQEST register from CC2400. What I am trying to understand is while I can see enqueue function being called to push the access address , but cant figure out where is the current channel/frequency value getting enqueued. Where should I be looking? Is my understanding of the code flow correct?

mikeryan, 2019-01-10:
`cb_follow_le` is a legacy callback that is not used. You will have to make your changes inside `le_phy.c`, probably by adding a field to the struct `le_rx_t`, polling the register inside `le_DMA_IRQHandler`, and finally finding a way to add it to the data sent over USB inside `usb_enqueue_le`. The actual channel is stored inside the global `rf_channel` and is stored in the `le_rx_t` struct for each packet. This is the absolute frequency in MHz (example: 2426 for advertising channel 38).

mikeryan, 2019-01-13:
I'm closing this because I think the above should get you going in the right direction. If you need more advice feel free to reopen.

closed 2019-01-13 by mikeryan