greatscottgadgets/ubertooth #346: Ubertooth-btle following & interfer

capcom1, 2019-01-29:
### Steps to reproduce
1. Run ubertooth-btle -p -I
2. 
3.

### Expected behaviour
After some time, ubertooth-btle should hop along with the active connection and interfer with it.

### Actual behaviour
ubertooth-btle -p -I displays "LE Promisc - Access Address: xxxxxxxx" and gets stuck here.
If you run ubertooth-btle -axxxxxxxx (your acces adress from the previous command) and then run ubertooth-btle -p -I you get the following output (reduced by me for easier reading)

root@kali:~#ubertooth -p -I
systime=1548769313 freq=2440 addr=af9a8a9a delta_t=238.409 ms rssi=-29
0d 00 84 06 08
Data / AA af9a8a9a (valid) /  0 bytes
    Channel Index: 17
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 1  SN: 1  MD: 0

    Data:
    CRC:   84 06 08

--------------------
LE Promisc - CRC Init: 38733f

systime=1548769315 freq=2440 addr=af9a8a9a delta_t=828.474 ms rssi=-23
01 00 f1 0d 08
Data / AA af9a8a9a (valid) /  0 bytes
    Channel Index: 17
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 0  SN: 0  MD: 0

    Data:
    CRC:   f1 0d 08

--------------------
LE Promisc - Hop interval: 48.75 ms

systime=1548769327 freq=2404 addr=af9a8a9a delta_t=1267.479 ms rssi=-32
0d 00 84 06 08
Data / AA af9a8a9a (valid) /  0 bytes
    Channel Index: 0
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 1  SN: 1  MD: 0

    Data:
    CRC:   84 06 08

--------------------
LE Promisc - Hop increment: 15

systime=1548769327 freq=2406 addr=af9a8a9a delta_t=243.752 ms rssi=-78
0d 00 84 06 08
Data / AA af9a8a9a (valid) /  0 bytes
    Channel Index: 1
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 1  SN: 1  MD: 0

    Data:
    CRC:   84 06 08

After that it stopps and the connection is still running.
Yesterday for some reason the interference was succesfull and the connection terminated.

I thought the ubertooth should keep following the connection and interfer with every channel.

### Version information
Operating system: kali linux, raspbian and windows

Ubertooth tools version (ubertooth-rx -V): libubertooth 1.1 (2018-08-R1), libbtbb 1.0 (2018-06-R1)

Ubertooth firmware version (ubertooth-util -v): Firmware version: 2018-08-R1 (API:1.05)

mikeryan, 2019-01-30:
"Interfere" mode is broken on the latest firmware. Marking this as a bug.

The latest firmware in which this feature still works is 2017-03-R2, so you can downgrade your host tools and firmware to that version if you want to use it.

capcom1, 2019-01-31:
Thanks for the answer.
I tried the firmware version 2017-03-R2 pretty much the whole day and it seems that this version is not reliable too. 

The interference worked 3 times for me and then not anymore.

The device gets stuck at the same step right after:

.....
systime=1548947256 freq=2440 addr=506544e9 delta_t=565.154 ms rssi=-100
05 00 b2 ec c6
Data / AA 506544e9 (valid) /  0 bytes
    Channel Index: 17
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 1  SN: 0  MD: 0

    Data:
    CRC:   b2 ec c6

systime=1548947256 freq=2440 addr=506544e9 delta_t=0.400 ms rssi=-100
0d 00 14 e1 c6
Data / AA 506544e9 (valid) /  0 bytes
    Channel Index: 17
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 1  SN: 1  MD: 0

    Data:
    CRC:   14 e1 c6

--------------------
LE Promisc - Access Address: 506544e9

Here i have to manually set the Access Address with
"ubertooth-btle -a506544e9"
and then run the "ubertooth-btle -p -I" again. This gives me:

......
--------------------
LE Promisc - Hop interval: 48.75 ms

systime=1548947283 freq=2440 addr=506544e9 delta_t=1803.537 ms rssi=-106
09 00 c7 e7 c6
Data / AA 506544e9 (valid) /  0 bytes
    Channel Index: 17
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 0  SN: 1  MD: 0

    Data:
    CRC:   c7 e7 c6

systime=1548947283 freq=2404 addr=506544e9 delta_t=97.497 ms rssi=-27
09 00 c7 e7 c6
Data / AA 506544e9 (valid) /  0 bytes
    Channel Index: 0
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 0  SN: 1  MD: 0

    Data:
    CRC:   c7 e7 c6

--------------------
LE Promisc - Hop increment: 10

systime=1548947284 freq=2406 addr=506544e9 delta_t=1267.742 ms rssi=-18
01 00 61 ea c6
Data / AA 506544e9 (valid) /  0 bytes
    Channel Index: 1
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 0  SN: 0  MD: 0

    Data:
    CRC:   61 ea c6

In both cases (succesfull Inteference and not succesful) the ubertooth gets stuck here.
If i abort here with CTRL+C i sometimes get the error "libUSB Error: Resource busy:  (-6)"

I also have to manualy reset the ubertooth with "ubertooth-util -r" and sometimes even remove and plug in the ubertooth again to scan ("ubertooth-btle -p") again.

I will test some more Firmware Versions tomorrow and give a report here again.

One last Question: Might it be possible to adapt the btle Intefere mode to "normal" Bluetooth?

capcom1, 2019-03-12:
Hello again,
i kept testing the interfere command of the ubertooth and got successfull jamming of BLE devices in promiscuous mode. In detail: i connected a IoT device (Bluetooth Thermostat) with my raspberry pi and jammed it successfully so it disconnected from the pi.

Then i tried the same with smartphone - thermostat.
However, in this case the ubertooth wasn't even able to detect the Access Address.
I first thought the ubertooth detects a lot of different Access Addresses so i changed the # of stored AA from 32 to 128 (in bluetooth_rxtx.c) but this doesn't solve the problem.

Any idea why the ubertooth can't detect the AA between the Phone and the Thermostat?