greatscottgadgets/ubertooth #349: Ubertooth one not working - Three solid leds: red, orange and light orange.

cldrn, 2019-02-02:
### Steps to reproduce
1. Got the device from Attify and plugged it in. 
2. Nothing gets detected on the computer (Windows or OS X hosts running Kali VM)
3. Three solid led lights are on as soon as you connect it: red, orange and light orange. 
4. DFU mode by shorting the 2 pins also does not work as the device is not detected.

Any ideas about what can be happening?

### Version information
**Operating system**:
Windows 10 and OS X as hosts, Kali VM 

**libbtbb version:**
Tried both:2018-08-R1 and 2017-03-R2
Bluez 5.50

[image]

GravesJake, 2019-02-11:
Which VM are you using? Do you see anything in the Windows device manager when in DFU mode? Which device ID are you passing though to your VM?

cldrn, 2019-02-15:
I'm using VMWARE fusion 10.1.5 and VMWare Workstation 12 Pro. i don't see anything with lsusb or in the windows device manager. Tried in both Windows and OSX hosts.

GravesJake, 2019-03-11:
Sorry for the delayed response, contact your re-seller for a replacement and we'll approve it.

closed 2019-03-11 by GravesJake