greatscottgadgets/ubertooth #351: Firmware built from recent source does not work

pdjstone, 2019-02-13:
### Steps to reproduce
1. Git clone master or use 2018-12-R1 release source
2. make bluetooth_rxtx firmware
3. ubertooth-dfu -d bluetooth_rxtx.dfu -r

### Expected behaviour
Firmware flashes and runs.

### Actual behaviour
The firmware seems to successfully build and flash. However after flashing the board shows a single green LED and doesn't appear over USB. I have to force the board into DFU mode via the pins in order to flash a working (precompiled) firmware. I've tried building both from git master and using the 2018-12-R1 release. I notice that my built version of bluetooth_rxtx.dfu is slightly larger (31061 bytes) than the 2018-12-R1 precompiled verson (30829 bytes)

### Version information
Ubuntu 18.04 with gcc-arm-none-eabi libnewlib-arm-none-eabi from standard repos.

libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

GravesJake, 2019-02-15:
Which compiler version are you using? I'm getting 30,709 bytes when building from f257505.

pdjstone, 2019-02-15:
gcc-arm-none-eabi -v gives:
 gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1)

pdjstone, 2019-02-15:
I've tested building the firmware with clean Ubuntu 18.04 and 16.04 VMs. The 16.04 VM built a working firmware using the ARM GCC toolchain from the stanard repos. So, to summarise:

- Ubuntu 16.04 / gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1) - working firmware
- Ubuntu 18.04 / gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1) - broken firmware

The working DFU file generated on 16.04 is 31348 bytes, so size doesn't seem to be causing an a problem.

GravesJake, 2019-03-11:
It appears that there's a bug with that particular gcc version, I'd recommend updating to a new version of gcc (8.2.0 in apt).

closed 2019-03-21 by GravesJake