greatscottgadgets/ubertooth #352: How do I Add "pcapbtbb" to the "logtypes=..." line in kismet.conf

bsb862, 2019-02-13:
Sorry, I am very new to this. I have my Ubertooth One and am not sure how to complete the step in 3rd party step for Kismet. 

The line I am referring to is : 

Add "pcapbtbb" to the "logtypes=..." line in kismet.conf

how do I do this? 

Thanks.

GravesJake, 2019-02-15:
Navigate to your kismet directory and `cd conf/`. Open `kismet.conf` and add "pcapbtbb" as a logtype (should be on line 278)

bsb862, 2019-02-15:
So is this correct? it was in the kismet_logging.conf file 

# Logging is enabled by type; plugins may add additional types.  The log types can be
# overridden on the command line with the '--log-types' argument.
#
# Built-in log types (plugins may add more, consult the documentation for plugins):
#   kismet      Unified log which can be turned into multiple types of data
#   pcapppi     Old-style pcap with PPI headers for signal and metadata.  Not as
#               flexible as the pcapng format.
#   pcapng      Pcap-NG (suitable for use with Wireshark and Tshark, as well as other
#               tools) which contains raw pcap data with interface tags.  See the
#               Kismet readme for methods to turn this into an old-style pcap log.
#
# By default, Kismet only enabled the unified 'kismet' log; the pcapng option is
# provided for special configurations as a legacy fallback mode.
log_types=kismet
log_types=pcapbtbb

bsb862, 2019-02-23:
Figured it out! I was on WSL Kali Linux.. Then I partitioned and made a bootable Kali.. THEN I just decided to go to Ubuntu because I was still having random library issues and stuff and now I've got it!

closed 2019-02-23 by bsb862