greatscottgadgets/ubertooth #353: libusb_init failed (got 1.0?)  any idea what I am missing here?

bsb862, 2019-02-13:
I am trying to connect the Ubertooth for the first time. Any idea what the "libusb_init failed (got 1.0?) means? 

### Actual behaviour
bouchedag@Bouche-Dag:~/kismet-2013-03-R1b$ ubertooth-rx -z
libusb_init failed (got 1.0?)
ubertooth-rx - passive Classic Bluetooth discovery/decode

Example usage:
        ubertooth-rx -- sniff for all LAPs
        ubertooth-rx -l <lap> -- calculate UAP for a given LAP
        ubertooth-rx -l <lap> -u <uap> -- calculate clock and follow piconet
        ubertooth-rx -z -t 20 -- survey mode: discover all LAPs+UAPs for 20 seconds

Major modes:
        -l <LAP> to decode (6 hex) - if not specified sniff all LAPs
        -u <UAP> to decode (2 hex) - if not specified calculate UAP (requires LAP)
        -z Survey mode - discover and list piconets (implies -s, interrupt with ctrl-C)
        -i <filename> input file - if not specified use Ubertooth for live capture

Configuration:
        -c <BT Channel> set a fixed bluetooth channel [Default: 39]
        -e max_ac_errors (default: 2, range: 0-4)
        -t <SECONDS> sniff timeout - 0 means no timeout [Default: 0]

Output options:
        -r<filename> capture packets to PcapNG file
        -q<filename> capture packets to PCAP file
        -d<filename> dump packets to binary file

Miscellaneous:
        -V print version information
        -U <0-7> set ubertooth device to use
bouchedag@Bouche-Dag:~/kismet-2013-03-R1b$ ubertooth-rx -V
libubertooth 1.1 (2018-08-R1), libbtbb 1.0 (git-6c7b9ff)

### Version information
**Operating system**:
Kali Linux (through windows 10 subsystem for Linux (WSL))
**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```

GravesJake, 2019-02-15:
Can you run the commands `ubertooth-rx -V` and `ubertooth-util -v` and paste the output of both here?

GravesJake, 2019-02-15:
Have you tried installing libusb? Instructions can be found in the [Build Guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide)

bsb862, 2019-02-15:
Hey Jake thanks for responding

bouchedag@Bouche-Dag:~/kismet-2013-03-R1b$ ubertooth-rx -V
libubertooth 1.1 (2018-08-R1), libbtbb 1.0 (git-6c7b9ff)
bouchedag@Bouche-Dag:~/kismet-2013-03-R1b$ ubertooth-util -v
libusb_init failed (got 1.0?)
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One

Common options:
        -v get firmware revision number
        -V get compile info
        -I identify ubertooth device by flashing all LEDs
        -d[0-1] get/set all LEDs
        -l[0-1] get/set USR LED
        -S stop current operation
        -r full reset
        -U<0-7> set ubertooth device to use
        -N print total number of Uberteeth and exit

Radio options:
        -a[0-7] get/set power amplifier level
        -c[2400-2483] get/set channel in MHz
        -C[0-78] get/set channel
        -q[1-225 (RSSI threshold)] start LED spectrum analyzer
        -t intitiate continuous transmit test
        -z set squelch level

Range test:
        -e start repeater mode
        -m display range test result
        -n initiate range test

Miscellaneous:
        -f activate flash programming (DFU) mode
        -i activate In-System Programming (ISP) mode
        -b get hardware board id number
        -p get microcontroller Part ID
        -s get microcontroller serial number

GravesJake, 2019-02-15:
Sorry, my comments are backwards. Try installing libusb before trying to use any ubertooth tools. How to do so can be found in the [Build Guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide) as it's OS specific.

bsb862, 2019-02-15:
bouchedag@Bouche-Dag:~$ sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev \
> pkg-config libpcap-dev python-numpy python-pyside python-qt4
Reading package lists... Done
Building dependency tree
Reading state information... Done
cmake is already the newest version (3.13.2-1).
g++ is already the newest version (4:8.2.0-2).
gcc is already the newest version (4:8.2.0-2).
libbluetooth-dev is already the newest version (5.50-1).
libpcap-dev is already the newest version (1.8.1-6).
libusb-1.0-0-dev is already the newest version (2:1.0.22-2).
make is already the newest version (4.2.1-1.2).
pkg-config is already the newest version (0.29-6).
python-numpy is already the newest version (1:1.16.1-1).
python-pyside is already the newest version (1.2.2+source1-3).
python-qt4 is already the newest version (4.12.1+dfsg-2+b1).
The following packages were automatically installed and are no longer required:
  libicu-le-hb0 libicu60
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

bsb862, 2019-02-15:
I'm fairly certain I have installed everything correctly, but having trouble getting started :/

bsb862, 2019-02-15:
I mean I'm obviously missing something lol but yeah having trouble figuring it out.

bsb862, 2019-02-18:
Hello, any ideas of what I could be missing? Thank you!

GravesJake, 2019-02-18:
WSL does not appear to have USB support at the moment. You might have to try running Linux directly or in a VM.

bsb862, 2019-02-20:
Ah.. Yes. That would be a problem, wouldn't it? Thank you!

closed 2019-02-22 by bsb862