greatscottgadgets/ubertooth #354: Repeated failed installs for ubertooth on Kali

Crypto-Cowboy, 2019-02-23:
Can anyone refer me to a current tutorial or working steps? I have tried every step by step tutorial I could find , I copied and pasted all commands. I am new to Kali and linux any help would be greatly appreciated.

bsb862, 2019-02-25:
I'm in the same boat as you. I assumed Kali would be ideal for Ubertooth, but it turned out that just installing and running through the steps on Ubuntu I had significantly less issues.

reallybigcatheter, 2019-03-08:
I have also been trying unsuccessfully to install under Kali - but here are some things I have found.

The Build Guide for Release 2018-12-R1, found here:  

https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

do not work - specifically the command to install all these packages:

sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev pkg-config libpcap-dev python-numpy python-pyside python-qt4

Several of these packages have been updated, so will not install.   

I had to download and install cmake from here:  https://cmake.org/install/

Running:

dpkg -l 'pkgname*' yields available packages for your platform.  That's a dash lowercase 'L' after dpkg, and enclose your search criteria within single quotes.  I used the asterisk as a wildcard to yield all possible name variants - then selected the latest.  for example:

libusb is libusb-0.1-4

libpcap-dev is actually libpcap0.8 

The others seem fine.

This *should get you to the point of installing the libbtb libraries.

I'm seeing issues with references to pkg-config when I try to compile Ubertooth tools, so I'll be opening a new issue report.

Hope this helps you.

dominicgs, 2019-03-11:
`dpkg -l` lists the currently installed packaged, for available packages use `apt search` or `apt-get search`.

I'm not sure which version of Kali you are using, but everything except python-pyside and python-qt4 should certainly be available from the apt repositories.

mikeryan, 2020-12-05:
This will be fixed when #383 is fixed.

closed 2020-12-12 by mikeryan