greatscottgadgets/ubertooth #355: Package python-pyside is not available

Crypto-Cowboy, 2019-02-24:
### Steps to reproduce
1. followed first step that is the same for all tutorials I have found
# sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev \
> pkg-config libpcap-dev python-numpy python-pyside python-qt4
2.
3.

### Expected behaviour
Tell us what you expect should happen
should verify versions and install

### Actual behaviour
Tell us what happens instead
After typing the first line command I get the following
# sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev \
> pkg-config libpcap-dev python-numpy python-pyside python-qt4
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python-pyside is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-pyside' has no installation candidate

### Version information
**Operating system**:
Kali 2018.4 amd64 was my iso

**Ubertooth tools version (ubertooth-rx -V):**
2018-12

**libbtbb version:**
2018-12

**Ubertooth firmware version (ubertooth-util -v):**
command not found

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

bsb862, 2019-02-25:
I was having the same error with Kali. I used Ubuntu and for whatever reason it started working.

Crypto-Cowboy, 2019-02-25:
I just tried it on Ubuntu and I got the same exact error. I also cant use specan-ui
root@kali:~# ubertooth-specan-ui 
bash: ubertooth-specan-ui: command not found

GravesJake, 2019-03-02:
Have you tried installing pyside with pip?

Crypto-Cowboy, 2019-03-09:
I have another issue open that is relater and will keep the thread going their

closed 2019-03-09 by Crypto-Cowboy