greatscottgadgets/ubertooth #356: Can someone walk me through enabling the kismet ubertooth plugin?

bsb862, 2019-02-27:
I am stuck here because I am new to linux. I don't know what I am supposed to be copying where or how to mess with the directories. Is this saying that I need to go into the kismet file and find an ubertooth plugin and copy it into the ubertooth file? I'm not really understanding what I'm supposed to do. Also how do I make sure the plugin shows up in kismet once the files and stuff are in the appropriate places? 

### Steps to reproduce
"By default, Kismet-Ubertooth expects the Kismet source to be in
	/usr/src/kismet; this can be overridden by setting the KIS_SRC_DIR
	environment variable:

		cd plugin-ubertooth
		KIS_SRC_DIR=/home/foo/src/kismet make

4.  Installing

	Kismet plugins may be installed system-wide in the plugins directory (by
	default, /usr/local/lib/kismet/) or in the users home directory
	(~/.kismet/plugins).

	To install in the system-wide directory:

        cd plugin-ubertooth
        KIS_SRC_DIR=/home/foo/src/kismet make install

    Plugins can be installed in the current users home directory with:

        cd plugin-ubertooth
        make userinstall
"

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
It's not clear to me which files need to be where and I get errors like "no rule to make userinstall"  

### Version information
**Operating system**:
Ubuntu 12.10

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (2018-08-R1),

**libbtbb version:**
 libbtbb 1.0 (2018-06-R1)
**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2018-08-R1 (API:1.05)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Kismet 2013-03-R0
### Output
```
Insert any commandline or build output here
```

bsb862, 2019-02-27:
Sorry my mystake, the operating system is Ubuntu 18.04.2

GravesJake, 2019-03-02:
Kismet plugins are currently deprecated due to many recent changes to Kismet.

closed 2019-05-21 by GravesJake