greatscottgadgets/ubertooth #358: Unable to install Ubertooth tools on Kali

reallybigcatheter, 2019-03-08:
Following the steps below from https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

wget https://github.com/greatscottgadgets/libbtbb/archive/2018-12-R1.tar.gz -O libbtbb-2018-12-R1.tar.gz
works
tar -xf libbtbb-2018-12-R1.tar.gz
works

cd libbtbb-2018-12-R1
works

mkdir build
works

cd build
cmake ..
*boom*

I'm seeing errors relating to pkgconfig and libusb

See output at bottom of this post.

### Version information
**Operating system**: Debian 4.19.20-1kali1 (2019-02-14)

**Ubertooth tools version (ubertooth-rx -V): ubertooth-2018-12-R1 <not yet installed>

**libbtbb version: 2018-12-R1

**Ubertooth firmware version (ubertooth-util -v): unknown

### Output
cmake ..
CMake Warning (dev) at CMakeLists.txt:28 (set):
  implicitly converting 'INT' to 'STRING' type.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:29 (set):
  implicitly converting 'INT' to 'STRING' type.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Setting version string 2018-12-R1
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Could NOT find LIBUSB (missing:  LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR) 
-- Building shared library
-- Setting udev rule group to - plugdev
-- Ubertooth udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /root/Downloads/ubertooth-2018-12-R1/host/libubertooth/src
   used as include directory in directory /root/Downloads/ubertooth-2018-12-R1/host/libubertooth/src

<snip>

LIBUSB_LIBRARIES (ADVANCED)
    linked by target "ubertooth" in directory /root/Downloads/ubertooth-2018-12-R1/host/libubertooth/src
    linked by target "ubertooth-specan" in directory /root/Downloads/ubertooth-2018-12-R1/host/ubertooth-tools/src
    linked by target "ubertooth-util" in directory /root/Downloads/ubertooth-2018-12-R1/host/ubertooth-tools/src

<snip>

-- Configuring incomplete, errors occurred!
See also "/root/Downloads/ubertooth-2018-12-R1/host/build/CMakeFiles/CMakeOutput.log".

Confirmed libusb is installed:

dpkg -s libusb-0.1-4
Package: libusb-0.1-4
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 58
Maintainer: Aurelien Jarno <aurel32@debian.org>
Architecture: amd64
Multi-Arch: same
Source: libusb
Version: 2:0.1.12-32
Replaces: libusb-dev (<< 2:0.1.12-25)
Depends: libc6 (>= 2.15)
Description: userspace USB programming library
 Library for programming USB applications without the knowledge
 of Linux kernel internals.
 .
 This package contains what you need to run programs that use this
 library.
Homepage: http://www.linux-usb.

Also attaching CMakeOutput.log
[CMakeOutput.log](https://github.com/greatscottgadgets/ubertooth/files/2947689/CMakeOutput.log)

GravesJake, 2019-03-11:
Try running the commands from the [Debian prerequisites section](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide#prerequisites):

```
sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev \
pkg-config libpcap-dev python-numpy python-pyside python-qt4
```

TheDolli, 2019-05-12:
I had the same problems with my Kali. Solved that by installing the libraries one by one. I do not know why, but in this way it simply makes problems. I did not occur these problems while installing on Ubuntu, however, am solving other issues I addressed in the new issue ticket #368

tommcq, 2020-04-14:
I had similar problems, in the end I stuck with Ubuntu; assuming/hoping it would be easier to install the user programs (like ubertooth-xxx) on Ubuntu, then the libraries on Kali. So far it has been.

mikeryan, 2020-12-05:
This will be fixed when #383 is fixed.

closed 2020-12-12 by mikeryan