greatscottgadgets/ubertooth #363: Receiving malformed packets

ankitv1995, 2019-03-25:
### Steps to reproduce
I followed the procedure described to setup the Ubertooth device 

### Expected behaviour
When I ran the ubertooth-btle -f -c /tmp/pipe command I started getting packets in the Wireshark as PPI protocol as expected.

### Actual behaviour
When I tried to set the btle as payload protocol in the DLT USER table, I saw that all the broadcast packets displayed "[MALFORMED PACKET: BT Common]" message.
So, now whenever I'm trying to capture packets using Ubertooth I'm getting this message. 
I can see Ubertooth working fine in the specan UI. 

### Version information
**Operating system**:
Ubuntu 16.04.6 LTS
**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 
**libbtbb version:**
libbtbb 1.0
**Ubertooth firmware version (ubertooth-util -v):**
2018-12-R1 (API: 1.06)
### Output

[image]

mossmann, 2019-06-03:
Based on the fact that there are no repeated source addresses, my guess is that these packets were actually received malformed.  There appears to be one correctly formed packet in the output above.  Can you take a look at that one in detail?
What does the output of ubertooth-btle -f look like?

mikeryan, 2020-12-05:
Closing this issue as it's very old. Please reopen if you're still having problems.

closed 2020-12-05 by mikeryan