greatscottgadgets/ubertooth #365: bluetooth_rxtx.dfu: No such file or directory

Rogersbugs, 2019-04-11:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
I am trying to update the firmware by following these instructions.
First, grab the latest Ubertooth release. Then, extract the archive and change into directory ubertooth-one-firmware-bin. And then run ubertooth-dfu command like so:

$ ubertooth-dfu -d bluetooth_rxtx.dfu -r

### Actual behaviour
But when I run the command, I get bluetooth_rxtx.dfu: No such file or directory yet I have extracted all the downloaded files into ubertooth-one-firmware-bin.

### Version information
**Operating system**:
Ubuntu

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2017-03-R2 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

mikeryan, 2019-04-11:
Can you please confirm that you downloaded this file:

https://github.com/greatscottgadgets/ubertooth/releases/download/2018-12-R1/ubertooth-2018-12-R1.tar.xz

And please show the output of the command `ls`

mactavishmeng, 2019-06-18:
relative path to 'bluetooth_rxtx.dfu' is `ubertooth-2018-12-R1/ubertooth-one-firmware-bin/bluetooth_rxtx.dfu`. You may check it manually to make sure the file is existed

closed 2020-12-01 by mikeryan