greatscottgadgets/ubertooth #366: Unable to run ubertooth-specan-ui

BerlinLiebe, 2019-04-14:
### Steps to reproduce
1. I followed build guide for installing ubertooth and ubertooth tools and wireshark - all completed successfully
2. I followed the guide for updating the firmware - this completed successfully

### Expected behaviour
I want to now run "sudo ubertooth-specan-ui"

### Actual behaviour
It fails with the following output:

Traceback (most recent call last):
  File "/usr/local/bin/ubertooth-specan-ui", line 29, in <module>
    from PySide2 import QtCore, QtGui, QtWidgets
ImportError: No module named PySide2

### Version information
Kubuntu 18.04

**Ubertooth tools version (ubertooth-rx -V): libubertooth 1.1 (2018-12-R1), 

**libbtbb version: libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v): Firmware version: 2018-12-R1 (API:1.06)

GravesJake, 2019-04-24:
Have you tried installing pyside from apt?

BerlinLiebe, 2019-04-26:
I have now (not that the instructions said so!)

And it still doesnt work - same error message.

GravesJake, 2019-04-26:
Sorry I forgot to specify the version number, can you verify that you have pyside**2** installed?

sunliBella, 2019-06-22:
I have the same problem. Can I know your solution? Thank you.    
when i input:  sudo apt-get install python-pyside2                
E: No package python-pyside2 was found

GravesJake, 2019-06-24:
@sunliBella try `python-pyside`, the python3 version is called `python3-pyside`

hmerkOneMedical, 2019-07-01:
Installing Pyside2 with pip also did not work for me. In case anyone else is having this issue, run `brew install ubertooth` ([https://brewinstall.org/Install-ubertooth-on-Mac-with-Brew/](https://brewinstall.org/Install-ubertooth-on-Mac-with-Brew/)) and it works like a charm. Run specan-ui again, and it should work without problems.

helenamerk, 2019-07-01:
Installing Pyside2 with pip also didn't work for me. In case anyone else is having this issue, run `brew install ubertooth` ([https://brewinstall.org/Install-ubertooth-on-Mac-with-Brew/](https://brewinstall.org/Install-ubertooth-on-Mac-with-Brew/)) and it works like a charm. Then you can run specan-ui again, and it should work without problems.

freemansoft, 2019-07-28:
Receive message
> E: Unable to locate package python-pyside2
When running
> ubertooth-specan-ui
On Kali
> 2019.02
 
Fix
- [ x]  pip install PySide2

atifplus, 2020-04-17:
> Installing Pyside2 with pip also didn't work for me. In case anyone else is having this issue, run `brew install ubertooth` (https://brewinstall.org/Install-ubertooth-on-Mac-with-Brew/) and it works like a charm. Then you can run specan-ui again, and it should work without problems.

still getting the same error it is not work for me ( on Mac OS)

atifplus, 2020-04-17:
> > Installing Pyside2 with pip also didn't work for me. In case anyone else is having this issue, run `brew install ubertooth` (https://brewinstall.org/Install-ubertooth-on-Mac-with-Brew/) and it works like a charm. Then you can run specan-ui again, and it should work without problems.
> 
> still getting the same error it is not working for me ( on Mac OS)

I had to also do the pip install PySide2 after the brew install ubertooth. It is now working

changedsoul, 2020-05-09:
I am not able to install brew or pip as it seems I always get a response "Unable to locate package xxxx".
Can anyone help me get past this hurtle?
Fresh install of Ubuntu 20.04, running the command:
  `sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev pkg-config libpcap-dev python-numpy python-pyside python-qt4`
fails with:

> Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python-pyside
E: Unable to locate package python-qt4

mikeryan, 2020-12-05:
This will be fixed when #383 is fixed.

closed 2020-12-12 by mikeryan