greatscottgadgets/ubertooth #367: Ubertooth not listed as hci device on Kali VM

Datacrass, 2019-04-16:
Running 4.17.0-kali1-amd64 #1 SMP Debian 4.17.8-1kali1 (2018-07-24) x86_64 GNU/Linux and have installed ubertooth using Kali apt repository (apt-get install ubertooth) and installing firmware manually on the device. 

The VM is running on VMware Workstation 15.x whereas I have a similiar VM on VMWare workstation 10.x Each VM is a clone of one another (I use the base image across 2 different hosts).

On the OS X side (Fusion) the device shows normally under hciconfig -a. 

On the Windows VMWare Workstation Side, the device does NOT show under hciconfig -a.

lsusb shows the device: Bus 002 Device 008: ID 1d50:6002 OpenMoko, Inc.

Below is the output from the terminal commands run 
Ubertooth util output: ubertooth-util -v
_Firmware version: 2018-12-R1 (API:1.06)_

ubertooth-util -t
_Starting TX test_

ubertooth-scan -s
_error: Unable to find hci0 (-1)_

sudo hciconfig -a (nothing printed)

hcitool scan 
_Device is not available: No such device_

Expected behaviour
I would think that since it can SEE the device and pick up the firmware it would load as a device under the hci devices.

### Version information
**Operating system**:
4.17.0-kali1-amd64 #1 SMP Debian 4.17.8-1kali1 (2018-07-24) x86_64 GNU/Linux
**Ubertooth tools version (ubertooth-rx -V):**
ubertooth-rx -V
_libubertooth 1.1 (2018-12-R1),_ 
**libbtbb version:**
_libbtbb 1.0 (2018-06-R1)_
**Ubertooth firmware version (ubertooth-util -v):**
_Firmware version: 2018-12-R1 (API:1.06)_

mikeryan, 2019-04-16:
Ubertooth is not a Bluetooth dongle and won't show up under hciconfig. For more info about the Ubertooth commands and what you can do with it, refer to the manual pages:

https://github.com/greatscottgadgets/ubertooth/blob/5441b44989dfc40d4cc83933d47c7808b2671707/host/doc/ubertooth.md
https://github.com/greatscottgadgets/ubertooth/tree/5441b44989dfc40d4cc83933d47c7808b2671707/host/doc

closed 2019-04-16 by mikeryan

DuckSounds, 2023-04-14:
But then how is this error fixable?
└─$ ubertooth-scan   
error: Unable to find hci0 (-1)

doblekali, 2025-03-25:
> But then how is this error fixable? └─$ ubertooth-scan error: Unable to find hci0 (-1)

Were you able to solve the problem?