greatscottgadgets/ubertooth #368: Firmware update problems, Could not open Ubertooth device

TheDolli, 2019-05-11:
### Steps to reproduce
1. Installed via these steps [https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide](url)
2. Attempted for firmware update but was unable to do so.
3. 

### Expected behaviour
Being able to see my firmware version after running `ubertooth-util -v`

### Actual behaviour
I am unable to update firmware, nor enter bootloader mode manual by connecting pins 1 and 3. I can see leds flashing when I do so for short while, but after running `ubertooth-dfu -d bluetooth_rxtx.dfu -r` I still get 

> could not open Ubertooth device
> Unable to find Ubertooth

### Version information
**Operating system**: Running via Oracle VM with virtual OS Kali 2019 version. 

**Ubertooth tools version (ubertooth-rx -V):** 
libubertooth 1.1 (2018-12-R1)

**libbtbb version:**  libbtbb 1.0 (git-7860bda*)

**Ubertooth firmware version (ubertooth-util -v):** I cannot get firmware version of it. It cannot open the Ubertooth device.

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One

Common options:
	-v get firmware revision number
	-V get compile info
	-I identify ubertooth device by flashing all LEDs
	-d[0-1] get/set all LEDs
	-l[0-1] get/set USR LED
	-S stop current operation
	-r full reset
	-U<0-7> set ubertooth device to use
	-N print total number of Uberteeth and exit

Radio options:
	-a[0-7] get/set power amplifier level
	-c[2400-2483] get/set channel in MHz
	-C[0-78] get/set channel
	-q[1-225 (RSSI threshold)] start LED spectrum analyzer
	-t intitiate continuous transmit test
	-z set squelch level

Range test:
	-e start repeater mode
	-m display range test result
	-n initiate range test

Miscellaneous:
	-f activate flash programming (DFU) mode
	-i activate In-System Programming (ISP) mode
	-b get hardware board id number
	-p get microcontroller Part ID
	-s get microcontroller serial number

```

mossmann, 2019-06-03:
Please try running `ubertooth-dfu -d bluetooth_rxtx.dfu -r` as root. It is important to execute this command quickly (within a couple seconds) after using pins 1 and 3 to enter bootloader mode.

TheDolli, 2019-07-01:
I did try to do so. Eventually I did not discover why I had such problems with ubertooth, however, I discovered it has something to do with device I tried to run it on itself. I tried out various virtual OS, all of them had same issue in the very end: They were unable to see the ubertooth devica, no matter what I tried.
Therefore, I was unable to run `ubertooth-dfu -d bluetooth_rxtx.dfu -r`
After I decided to try ubertooth out on other devices I successfuly interacted with the device on raspberry pi, although different unrelated issues araised there, and then on MacOs where ubertooth worked without any issues. 
First laptop I tried ubertooth to run on is LenovoIdeaPad Yoga 3 14. I might try to discovered where the issues laid in future, however for now I switched to different device as it is more convenient.

mossmann, 2019-07-03:
Interesting.  It sounds like your firmware issue is resolved.  If you would like help getting the Ubertooth to work with a particular host platform, please open another issue.  (As a start, i suggest watching dmesg while plugging it in.)

closed 2019-07-03 by mossmann

Sonali433, 2022-05-29:
hi there, 

I have configured for ubertooth one Kali linux. But unfortunately, I am unable to contact with it. Getting the message; 

ubertooth-util -v                                                                                                                                               11 ⚙
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One

Common options:
	-v get firmware revision number
	-V get compile info
	-I identify ubertooth device by flashing all LEDs
	-d[0-1] get/set all LEDs
	-l[0-1] get/set USR LED
	-S stop current operation
	-r full reset
	-U<0-7> set ubertooth device to use
	-N print total number of Uberteeth and exit

Radio options:
	-a[0-7] get/set power amplifier level
	-c[2400-2483] get/set channel in MHz
	-C[0-78] get/set channel
	-q[1-225 (RSSI threshold)] start LED spectrum analyzer
	-t intitiate continuous transmit test
	-z set squelch level

Range test:
	-e start repeater mode
	-m display range test result
	-n initiate range test

Miscellaneous:
	-f activate flash programming (DFU) mode
	-i activate In-System Programming (ISP) mode
	-b get hardware board id number
	-p get microcontroller Part ID
	-s get microcontroller serial number
	-x xmas lights
------------------------------------------------------------------------
#  ubertooth-specan-ui                                                                                                                                        1 ⨯ 11 ⚙
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Could not open Ubertooth device
Failed to run:  ubertooth-specan -d - -l 2400 -u 2483 -U -1