greatscottgadgets/ubertooth #370: Will Ubertooth work with active Bluetooth RFID tags?

Russ76, 2019-06-11:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

GravesJake, 2019-06-21:
I don't have experience with RFID tags, but a good place to ask would be in the ubertooth IRC channel.

mikeryan, 2020-12-01:
Those are BLE, so yes!

closed 2020-12-01 by mikeryan