greatscottgadgets/ubertooth #372: Unable to sniff the Pairing Public Key between the BLE pairing progress

sunliBella, 2019-06-25:
### Steps to reproduce
1.Run in the terminal: ubertooth-btle  -f -c a.pcap
2.Set up two virtual devices--heart rate in LightBlue APP in iphone and ipad.
3.Open the Bluetooth and search the peripherals nearby in the app
4.Choose the pairing device, do the numeric comparison and complete pairing between the two heart rate virtual devices.
5.Repeat the above steps.
ps: I also repeat the pairing between the smart watch, for example, the Mi Band, but i can't get any smp data.

### Expected behaviour
The ubertooth could sniff the smp data package, pairing request,pairing response,pairing random,pairing dhkey check and pairing public key.

### Actual behaviour
In multiple tests, i can get the smp data, e.g. pairing response,pairing random,pairing dhkey check for several times. But i only capture the pairing public key only once. 
How can i get more pairing public key?
 Or what devices can i use in order to get more pairing public key? 

### Version information
**Operating system**:Oracle VM VirtualBox  Ubuntu 14.04.1 64bit
**Ubertooth tools version (ubertooth-rx -V):libubertooth 1.1(2018-12-R1)
**libbtbb version:**,libbtbb 1.0 (2018-06-R1)
**Ubertooth firmware version (ubertooth-util -v):**2018-12-R1(API:1.06)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```I really need help. Looking forward to answering as soon as possible. Thank you very much

sunliBella, 2019-06-25:
I really need your help. Looking forward to answering as soon as possible. Thank you very much

mikeryan, 2020-12-05:
If you're still having this problem, please reopen the issue.

closed 2020-12-05 by mikeryan

kenshinsamue, 2021-03-20:
Im having the same issue with the Dec 2020 firmware