greatscottgadgets/ubertooth #373: Ubertooth follow doesn't set the clock correctly

cathyz64, 2019-07-08:
In ubertooth-follow, it uses hci_read_clock_offset to get the hopping pattern of the slave device. However, the offset is right shifted by 2 so it shouldn't be clock += offset, it should be clock += offset<<2. But there is another problem, because usleep(10000) is called before that clock value is sent to the ubertooth. It would be better to create a connection to the ubertooth prior to reading the clock offset and then send clock + offset<<2 + delay immediately to the ubertooth. You can then delay and close out the connection. I have tested this method and it works perfectly for getting the correct slave clock and thus the correct hopping pattern.

GravesJake, 2019-07-09:
Feel free to open a pull request with this change!

mikeryan, 2020-12-01:
I second the request for a pull request @cathyz64

aerial64, 2020-12-01:
I did the pull request last summer. I think it made it through but I'm off on other things and haven't followed up.

cathyz64, 2020-12-01:
I believe I made the pull request last summer. I think it went through with a few build errors that may need to be resolved on the gsg side of things.

mikeryan, 2020-12-04:
Thanks, I missed that one. I will review the PR before our next release.

closed 2020-12-05 by mikeryan