greatscottgadgets/ubertooth #376: USB device not recognised

HeideFisher, 2019-08-23:
### Steps to reproduce
1. Inserting Ubertooth in Windows machine
2.
3.

### Actual behaviour
An error message showed 'The last USB device you connected to with this computer malfulnctioned and Windows does not recognise it.

### Version information
Windows 10:

I have downloaded the libbtbb-2012-10-R3 and ubertooth-2018-12-R1 into the kali linux operating system which is running on VMWare on my WIndow10 machine. My machine does not seem to read the ubertooth USB. 

I have also tried to insert the Ubertooth USB directly into my Raspberry Pi 3 (after downloading libbtbb-2017-03-R2 and ubertooth-2017-03-R2) and again it was not recognised.
If I run ubertooth-util -v the error message is the same on both machines 'could not open Ubertooth device'

EMCP, 2019-09-09:
I think you may need to check your devices LEDs.. because mine has been working fine off a RPi3

HeideFisher, 2019-09-10:
I thought it would work off a RPi3. When I plug the Ubertooth into my Pi the lights do not even go on. Maybe my device is broken ??

EMCP, 2019-09-10:
correct, if it isn't even getting power to the LEDs thats a bad sign.. I don't know how you'd debug that though and you may want to research that part

luzat, 2020-06-15:
I  received a device which was missing Q1 and R15. In this case, many things (testing, flashing through FTDI programmer) seemed to work, except USB. Checking whether all components are in their place and going through firmware/assembly_test and usb_test might be helpful (if interested in fixing it instead of having it replaced).

BG7YWL, 2020-07-05:
> I received a device which was missing Q1 and R15. In this case, many things (testing, flashing through FTDI programmer) seemed to work, except USB. Checking whether all components are in their place [...]

I tried assembly_set and usb_test and found that the USB light could not be turned on, please tell me where to check the circuit.
[image]

mikeryan, 2020-12-01:
@HeideFisher it sounds like your device may be defective. Where did you buy it from?

closed 2021-05-28 by straithe