greatscottgadgets/ubertooth #377: Unable to sniff data packets.

r00tb3, 2019-08-25:
### Steps to reproduce
1. sudo hcitool scan
2. ubertooth-btle -f -t B8:27:EB:XX:XX:XX -c /tmp/pipe
3. sudo wireshark

### Expected behaviour
I should be able to see pairing packets and the data packets that are being sent for opening or closing door in this example

### Actual behaviour
I'm able to see only the advertisement packet (ADV_INT), scan request(SCAN_REQ) and scan response(SCAN_RESP) packets once my smartphone pairs with the device through its app and I send door open and close commands they aren't being sniffed.

### Version information
**Operating system**: 16.04.1-Ubuntu

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 1.1 (2018-12-R1)

**libbtbb version:** libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):** 2018-12-R1 (API:1.06)

Why I'm not able to sniff the data packets?

I want to know the characteristic values which I can replay and make the doors open/close.

Regards.

shoulewoba, 2019-09-21:
I have exactly same situation.

CthUlhUzzz, 2019-10-15:
The same problem

mikeryan, 2020-12-01:
`hcitool scan` gives you scan results for classic Bluetooth devices. `ubertooth-btle` is used for sniffing BLE.

This may be solved with #427

closed 2020-12-01 by mikeryan