greatscottgadgets/ubertooth #379: Another libUSB Error: Operation timed out:  (-7)

00Roeder, 2019-09-18:
### Steps to reproduce
1. ubertooth-btle -n -c /tmp/pipe

### Expected behaviour
dumping into the pipe

### Actual behaviour
libUSB Error: Operation timed out:  (-7)
after 10-60 sec (varies)

### Version information
**Operating system**:
Kali 2019
**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2018-12-R1 (API:1.06)

freqyXin, 2019-09-30:
I'm having identical issues. Any luck finding a work around or solution?

neomorphus, 2019-11-29:
i am facing the same issue, does anyone find a solution?

tranzmatt, 2020-02-12:
I'm having an issue with mine, but with kismet inside a docker.  I can run `ubertooth-rx ok`, but `kismet -c ubertooth` borks with a `libUSB error (-4)`.  I was wondering if these were related.

cdrdzsilva, 2021-01-04:
Anyone was able to find the solution for this?

closed 2021-07-18 by straithe