greatscottgadgets/ubertooth #381: Ubertooth won't enter DFU

huntergregal, 2019-10-07:
I have an ubertooth i got a few years ago (still new in package) from Defcon. has `r161-p9` code on it.

Any tool i try to use tells me that it cannot open the ubertooth device. So I figure i need to short pins 1/3 and enter DFU to firmware upgrade it.

...Only no matter what i used to bridge the pins the 2 solid green lights stay on when i plug it and and lsusb/dmesg has no change. It just appears this ubertooth will NOT go into DFU mode.

I'm sure I'm bridging the correct pins as per the photo on the wiki.

Any ideas? I'd assume it's dead if it wasn't for the 2 solid green lights signalling it has power. (RSTLED and 1V8 LED)

https://github.com/greatscottgadgets/ubertooth/wiki/Ubertooth-One#pins-and-leds

huntergregal, 2019-10-07:
In addition is there something I could do to flash it manually over jtag or another interface?

mikeryan, 2019-10-07:
What kind of device are you plugging the Ubertooth into? On some OS X machines I've seen them being totally unwilling to enumerate or boot. Can you try in a Windows or Linux machine if you haven't yet?

huntergregal, 2019-10-07:
this is laptop running bare metal ubuntu 18. I have access to windows machine as well but didn't feel like playing  the vmware device pass through game

mikeryan, 2019-10-07:
This sounds like a problem with the bootloader. You can use a [pogoprog](https://github.com/greatscottgadgets/ubertooth/tree/master/hardware/pogoprog) to program it if you want to build some hardware. There is also an unpopulated standard ARM SWD header on top of the board.

Paging @mossmann who can help rule out whether this is a hardware problem and/or give you some other options.

huntergregal, 2019-10-07:
I'm pretty positive it's hardware at this point. I had a theory it could be a bad ground pin and was playing around trying to bridge pin 3 to any ground and while using the usb connector as a ground i got into dfu. of course i wasn't expected that to work and missed my flash opportunity and now i cant get back into dfu :(

It's "safe" to perma bridge pin 3 right? like to always boot into dfu first? I'm thinking ive soldering it to help. Seems like it may actually be a case of a wonky pin 3

mikeryan, 2019-10-07:
It is safe to permanently short pin 3 to ground via pin 1. It will boot into DFU for 5 seconds and then boot into normal mode.

huntergregal, 2019-10-07:
so i put a 6 pronged head on the expand header and used a jumper for pins 1/3 - no dice. 

Before this the annular ring (?) on the top side pin 3 hole was chipping off. Now im concerned that may be related to the issues. I'm pretty dead set on fixing this now -- ill probably pull the head back off and play with this a bit. I'd really like to get it working for some new security projects .

Thanks for your help. I'll go ahead and close this issue and open another one if i need to as things progress.

closed 2019-10-07 by huntergregal

huntergregal, 2019-10-07:
follow up - i pulled up the kicad file and found the expand pin 3 connects to pin 44 on the chip. tried grounding just that pin and nothing is happening when i plug it in. (I know it's the right pin because i confirmed the usb led pin does in fact turn on the usb led when i ground it). So im gonna assume it's dead. 

I think i'll bite the bullet and get the hack rf