greatscottgadgets/ubertooth #382: Could not open Ubertooth device ('Failed to run: ', 'ubertooth-specan -d - -l 2400 -u 2483 -U -1')

AntonioDeterville, 2019-10-11:
### Steps to reproduce
1. Installed all drivers and Libraries for Ubertooth successfully 
2. Tried to run "ubertooth-specan-ui" command 
3. Recevied the following Error:
Could not open Ubertooth device ('Failed to run: ', 'ubertooth-specan -d - -l 2400 -u 2483 -U -1')

### Expected behaviour
Ideally, the Ubertooth Spectrum UI should open up and give an graphical view of the signals the Ubertooth should be picking up. 

### Actual behaviour
I receive the error I described before: Could not open Ubertooth device ('Failed to run: ', 'ubertooth-specan -d - -l 2400 -u 2483 -U -1')

### Version information
**Operating system**: Using Virtual machine version 6.0.12
Running Ubuntu 18.04.3 LTS

**Ubertooth tools version (ubertooth-rx -V):** ubertooth-2018-12-R1

**libbtbb version:** libbtbb-2018-12-R1

**Ubertooth firmware version (ubertooth-util -v):** Honestly not sure which firmware, it should be the lastest one since i followed the Build guide on the GitHub Wiki. 

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```

I have tried adding my user account to the "vboxusers" group, and have already allowed the ubertooth device through the filter in Virtual Box. I also tried updating all the drivers using sudo apt-get upgrade, and tried reinstalling the libbtbb library.  I am new to Linux, so any guidance on how to solve this issue would be deeply appreciated. When i plug in the Ubertooth, the Host System (Windows 10) recognized the device normally, but the second I start the VM, the red LED on the Ubertooth goes off and the system no longer recognizes it.

AntonioDeterville, 2019-10-11:
Sorry, New to posting here, just adding the part that was in the output window:

I have tried adding my user account to the "vboxusers" group, and have already allowed the ubertooth device through the filter in Virtual Box. I also tried updating all the drivers using sudo apt-get upgrade, and tried reinstalling the libbtbb library.  I am new to Linux, so any guidance on how to solve this issue would be deeply appreciated. When i plug in the Ubertooth, the Host System (Windows 10) recognized the device normally, but the second I start the VM, the red LED on the Ubertooth goes off and the system no longer recognizes it.

mikeryan, 2019-10-11:
Please paste the output of ubertooth-util -v and lsusb

AntonioDeterville, 2019-10-12:
Good Afternoon, thank you for the prompt response, appreciated. Below is the output of ubertooth-util -v (doesn't seem to bringing up the version number though):
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One

Common options:
	-v get firmware revision number
	-V get compile info
	-I identify ubertooth device by flashing all LEDs
	-d[0-1] get/set all LEDs
	-l[0-1] get/set USR LED
	-S stop current operation
	-r full reset
	-U<0-7> set ubertooth device to use
	-N print total number of Uberteeth and exit

Radio options:
	-a[0-7] get/set power amplifier level
	-c[2400-2483] get/set channel in MHz
	-C[0-78] get/set channel
	-q[1-225 (RSSI threshold)] start LED spectrum analyzer
	-t intitiate continuous transmit test
	-z set squelch level

Range test:
	-e start repeater mode
	-m display range test result
	-n initiate range test

Miscellaneous:
	-f activate flash programming (DFU) mode
	-i activate In-System Programming (ISP) mode
	-b get hardware board id number
	-p get microcontroller Part ID
	-s get microcontroller serial number

Below is the output for lsusb:
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Hope this clarifies my problem

AntonioDeterville, 2019-10-23:
Good Night, after much trial and error, I found a website that i followed the steps to get the module working on my Linux Ubuntu Machine. Leaving the link in this comment for anyone who encounters the same problem. It seems that this was an issue with refreshing the libraries for the Ubertooth, and also the firmware running may be too outdated. This website gives step by step instructions on how to fix it. Hope this helps someone else one day. 
http://installfights.blogspot.com/2016/07/install-and-configure-ubertooth-one.html

closed 2019-10-23 by AntonioDeterville