greatscottgadgets/ubertooth #383: is ubertooth python3 compatibly?

sandrotosi, 2019-10-22:
Hello,
i couldnt find anywhere if ubertooth is python3 compatible.

In debian we are trying to remove python2 and ubertooth is one of the only 2 packages keeping pyside2 in the archive, so it would be extremely welcome if we could have a py3k compatible version of ubertooth in the near future.

thanks for considering!

mikeryan, 2019-10-22:
Looking into this right now. Is there a timeline by which this change
should occur? Also, are you the Ubertooth package maintainer on Debian?

sandrotosi, 2019-10-22:
thanks a lot Mike, much appreciated! as it always goes, the sooner the better :)
I'm not Ubertooth maintainer for debian, i'm part of a larger group of people working to remove python2 from the distribution

rubund, 2019-12-23:
I'm maintaining ubertooth in Debian. It has unfortunately been automatically removed from Debian testing now because of lack of Python3 support. Any timeline for Python3 support, or should I try to patch it myself?

mnhauke, 2019-12-25:
I'm maintaining ubertooth in openSUSE. Ubertooth was also dropped from openSUSE Tumbleweed some time ago because of the missing python3 support :/

mikeryan, 2020-12-05:
@rubund are you still supporting the package for Debian? Can it be re-included at this point?

closed 2020-12-05 by mikeryan

rubund, 2020-12-06:
@mikeryan  It was only outside Debian testing between 2019-12-15 and 2019-12-23 before I uploaded it without specan-ui (https://tracker.debian.org/news/1088591/accepted-ubertooth-201812r1-3-source-into-unstable/) . Then I did a quick fix for specan-ui and uploaded that 2020-03-28 (https://salsa.debian.org/rubund/ubertooth/-/blob/master/debian/patches/0007-Patch-to-run-in-Python-3.patch). Apparently I forgot about this ticket and forwarding the patch to you.

mikeryan, 2020-12-06:
@rubund The project dependencies changed as you can see on the [installation wiki page](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide). No longer requires libpcap or pyside2 but does require python3-numpy and python3-qtpy.

Is there a better way to reach you about coordinating these changes? You can email me at mike@ice9.us

commit 7388cebb (2020-12-05): python: upgrade to python 3

Fixes #383 #354 #358 and #366