greatscottgadgets/ubertooth #387: How to set advertizing data for ubertooth-btle?

StevenRudenko, 2019-11-13:
### Steps to reproduce
1. `ubertooth-btle -s00:11:22:33:44:55`

### Expected behaviour
Ubertooth device is scanabled within address `00:11:22:33:44:55`. Advertising data is readable

### Actual behaviour
Ubertooth device is no visible while scanning

### Version information
**Operating system**: MacOS 10.14.6 (18G95)

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:** 2018-12-R1?

**Ubertooth firmware version (ubertooth-util -v):** Firmware version: 2018-12-R1 (API:1.06)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
 ubertooth-btle -s00:11:22:33:44:55
```

nishant412, 2019-11-15:
I am facing the same issue. I have the same ubertooth firmware and tools versions as above, just using it on a different OS (Ubuntu 18.04). When I issue the fake advertising packets command with -s flag, I see the TXLED rapidly and continuously blinking but it doesn't actually transmit advertisements. 

To check for advertisements, I used the nrfconnect app on my Android phone, and also just in case checked with a USRP for any transmissions on the three advertising channels. Couldn't see any advertisements.

Thanks 
Nishant

evanslify, 2019-11-22:
Check `ubertooth-btle.c`. The data you wanted to change is in `uint8_t adv_data[]`. 
Note that you cannot exceed 21 bytes - https://github.com/greatscottgadgets/ubertooth/issues/350

StevenRudenko, 2019-11-22:
@evanslify thank, but problem is that device is not advertising itself at all. do you know how to fix that?

evanslify, 2019-11-22:
Are you listening on the right channel?

zhaohuansi, 2019-11-27:
Chinese？

closed 2020-12-01 by mikeryan