greatscottgadgets/ubertooth #390: Ubertooth-rx command says to update firmware

sameeshem, 2019-11-28:
### Steps to reproduce
1. Enter ubertooth-rx command
2.
3.

### Expected behaviour
Tell us what you expect should happen

i have purchased ubertooth from nooelec and followed https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide for installing host software. After installation when i tried ubertooth-rx command.i am getting following error

Ubertooth API version 1.01 found, libubertooth 1.1 requires 1.06.
Please upgrade to latest released firmware.
See: https://github.com/greatscottgadgets/ubertooth/wiki/Firmware

### Actual behaviour
Tell us what happens instead

Ubertooth API version 1.01 found, libubertooth 1.1 requires 1.06.
Please upgrade to latest released firmware.
See: https://github.com/greatscottgadgets/ubertooth/wiki/Firmware

### Version information
**Operating system**:
   Ubuntu 14.04
**Ubertooth tools version (ubertooth-rx -V):**
   
libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:**
libbtbb-2018-12-R1
**Ubertooth firmware version (ubertooth-util -v):**

Firmware version: 2015-10-R1 (API:1.01)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

mikeryan, 2019-11-28:
Follow these instructions to update the firmware:

https://github.com/greatscottgadgets/ubertooth/wiki/Firmware#how-to-update-firmware

sameeshem, 2019-11-29:
Thanks, its working!

closed 2020-12-01 by mikeryan

chuzairps, 2022-06-28:
hi there, 

I am facing issue 

# ubertooth-specan-ui                                                                                                                                          1 ⨯ 1 ⚙
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Could not open Ubertooth device
Failed to run:  ubertooth-specan -d - -l 2400 -u 2483 -U -1