greatscottgadgets/ubertooth #391: ubertooth-btle -s not transmitting advertisement data

nishant412, 2019-12-03:
### Steps to reproduce
1. Run the command ubertooth-btle -s22:44:66:88:aa:cc
2. Check BLE scanner app for address (I use nrf BLE scanner app)

### Expected behaviour
Ubertooth should transmit fake advertisement packets with the advertising address 22:44:66:88:aa:cc, that are detected on the app

### Actual behaviour
Don't see any advertisements from that address. I see the TXLED is continuously blinking (feebly) which makes me believe it is hitting the le_transmit() function in the code. But no advertisements are detected in the app.

### Version information
**Operating system**: MacOS 15.1

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:** libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):** Firmware version: 2018-12-R1 (API:1.06)

mikeryan, 2020-12-01:
Review this when #341 is merged.

mikeryan, 2020-12-12:
I've confirmed using two Uberteeth that transmit code is working on latest master.

closed 2020-12-12 by mikeryan