greatscottgadgets/ubertooth #395: Bluetooth MACs retrieved from ubertooth-scan are incorrect???!!

sn2n16, 2020-01-06:
### Steps to reproduce
1. ubertooth-scan -s
2.ubertooth-btle -t RetrievedBluetoothMACAddress
3.ubertooth-btle -f -c /tmp/pipe &
4. wireskark -k -i /tmp/pipe &

### Expected behaviour
Should follow the messages from RetrievedBluetoothMACAddress and show them in Wireshark. 

### Actual behaviour
Wireshark opens but does nothing. It doesn't have any traffic. Either doesn't follow the connection, or nothing is being transmitted. I think the static RetrievedBluetoothMACAddress, retrieved from ubertooth-scan cannot be used to follow the packets in ubertooth-btle -f. How can I find the advertising Bluetooth MAC Address of the device? Does the RetrievedBluetoothMACAddress of my iPhone 8 change, as soon as I start using ubertooth-btle? If so, how can I find and track/follow the new iPhone 8 MAC Address?

### Version information
**Operating system**:

**Ubertooth tools version (ubertooth-rx -V):**
ubertooth 2018-12-R1 (root@kaliLinux) Fri 13 Dec 16:24:03 GMT 2019
**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**
2018-12-R1 (API:1.06)
If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```Wireshark opens but does nothing. If I do ubertooth-btle -f, it works correctly but most Bluetooth MAC Addresses, shown in Wireshark under "Source" are different...
Insert any commandline or build output here
```

mikeryan, 2020-01-06:
`ubertooth-scan` looks for classic Bluetooth devices, while `ubertooth-btle` looks for BLE devices. Even if you are able to identify a partial address using `ubertooth-scan` it will not be useful for tracking your device via BLE.

iPhones use Resolvable Private Addresses (RPAs) when advertising, which change periodically. Even if you are able to identify your own device's BD ADDR, it will change a few minutes later. There is no way to track a device long-term passively, as this is what RPAs are designed to prevent. You will have to pair with the device and recover its IRK to resolve the private addresses to public. That is out of the scope of the Ubertooth project.

sn2n16, 2020-01-06:
Thank you for reply. I wasn't aware of that. If the iPhone is connected and transmitting, I should be able to see the packages through ubertooth-btle -f right? It would sniff all the Bluetooth devices, then just filter it in Wireshark using filters? But how will I know which device is my iPhone?

mikeryan, 2020-01-06:
Luck of the draw. `ubertooth-btle -f` will follow the first connection it observes. It might not "see" your connection (there is a 33% chance), and it might be following another connection while yours is established.

There's no way to tell with certainty during a connection which devices are communicating. You have to make a guess based on the contents of any captured packets.

closed 2020-12-01 by mikeryan