greatscottgadgets/ubertooth #396: Issues installing on ARM + Ubuntu

EdwinCarlJames, 2020-01-06:
### Steps to reproduce
1. Download tools from build guide
2. navigate to firmware directory 
3. make any of the firmware

### Expected behaviour
firmware should build

### Actual behaviour
"recipe for target 'bluetooth_rxtx.dfu' (or any other tool) failed"

### Version information
**Operating system**:
Ubuntu 18.04.3 LTS
Kernel: Linux 4.9.108 tegra
Architecture: arm64

**Ubertooth tools version (ubertooth-rx -V):**
2018-12-R1

**libbtbb version:**
2018-12-R1

**Ubertooth firmware version (ubertooth-util -v):**
not sure, can't run this command

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
-------- begin --------
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
  28808	   2565	  12598	  43971	   abc3	bluetooth_rxtx.elf

Creating DFU firmware file: bluetooth_rxtx.dfu
s bluetooth_rxtx.bin
make: s: Command not found
../common.mk:387: recipe for target 'bluetooth_rxtx.dfu' failed
make: [bluetooth_rxtx.dfu] Error 127 (ignored)

--------- Target Information ---------
ARM Model: cortex-m3
Board: UBERTOOTH_ONE
--------------------------------------

Size after:
   text	   data	    bss	    dec	    hex	filename
  28808	   2565	  12598	  43971	   abc3	bluetooth_rxtx.elf

-------- end --------

```

mikeryan, 2020-12-01:
Have you build and installed the host tools? What is the output of `which ubertooth-dfu`?

mikeryan, 2020-12-19:
Closing this due to inactivity. If you're still having this problem, please reopen the issue.

closed 2020-12-19 by mikeryan