greatscottgadgets/ubertooth #397: BLE-Ubertooth-wireshark - PPI malformed packets

sameeshem, 2020-01-07:
### Steps to reproduce
1.Capture PCAP packets using ubertooth-btle command
2.analyze PCAP packets through Wireshark
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead
Braodcast packet are shown correclty, but when data packet comes. i am getting a PPI malformed packet as shown in image

[image]

### Version information
**Operating system**:

**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

mikeryan, 2020-01-07:
Can you attach the PCAP file here. Please give the output of the tool requests above (`ubertooth-util -v` and `ubertooth-rx -V`)

sameeshem, 2020-01-08:
HI,
ubertooth-util -v --- Firmware version: 2018-12-R1 (API:1.06)
ubertooth-rx -V ------libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

[ble.zip](https://github.com/greatscottgadgets/ubertooth/files/4035953/ble.zip)

sameeshem, 2020-01-24:
Hi Mike

Any updates on issue?

ZakB-K, 2020-08-31:
Hi @sameeshem, I had a similar issue and following the instructions on the [wiki for troubleshooting Wireshark issues](https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark) fixed the issue for me. Specifically the note about `DLT=147`. I know it doesn't sound like it matches this problem exactly but it fixed it for me so I recommend giving it a try if you haven't already. Related to #61.

mikeryan, 2020-12-01:
@sameeshem can you try the latest master firmware and host code and see if this is still an issue?

mikeryan, 2020-12-19:
Closing this due to inactivity. Please reopen the issue if you're still having this problem.

closed 2020-12-19 by mikeryan