greatscottgadgets/ubertooth #398: Using CC2500 to replace CC2400

RonHreha, 2020-01-15:
We recently purchased an Ubertooth One from Great Scott Gadgets, and have been experimenting with it.  We are considering designing our own hardware based off the CC2500 as the CC2400 is not recommended for new designs (for the past 10 years it seems).  You answered a question from Hunyue Yau on 5/11/2012 indicating the CC2500 does not support 1 Mbps symbol rate required for Bluetooth.  I see from the datasheets for both products that yes, the CC2500 max data rate is half of the CC2400.  Do you believe we could ever get an Ubertooth like sniffer based off the CC2500 to work?  Could you please elaborate on the main issue we might run up against concerning the data rate difference?

mikeryan, 2020-01-15:
With only 500 kbit/sec you will simply be unable to capture BLE packets. The radio won't be able to sync on the preamble or sync word, and even if it could BLE devices would be sending two bits for each bit the CC2500 is able to receive.

If you're looking for a similar radio that's currently in production, consider the ADF7242 or perhaps a device from Nordic. The major caveat on those is that we haven't tested or validated them extensively for Bluetooth use. At least in the case of ADF7242 there shouldn't be anything in principle preventing it from working, but your mileage may vary. Plan to spend a lot of time with the datasheet.

tomsaul, 2020-02-19:
Take a look at https://www.amazon.com/gp/product/B01AL6Y9EK/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
There is actually a newer version based on the CC2640, but so far I have not seen anyone selling it with the 'Sniffer' software installed - but from what I read it should be usable there as well.

I've tested this device with the latest Kismet, and it seems to work well, with the only caveat being that it does not support an external antenna (lower power/range).

sultanqasim, 2020-02-27:
What do you want to sniff? The modern TI CC2652 and CC1352 are mostly a superset of the older receivers' capabilities, though it would require some experimentation to implement the sync on preamble feature used for access address discovery on the Ubertooth.

closed 2020-12-01 by mikeryan