greatscottgadgets/ubertooth #399: Migrating libBTBB with gr-bluetooth

tylerone1, 2020-02-27:
Hi all!
I bought HackRF SDR and also ubertooth.
I'm trying to write a simple BR/EDR sniffer using gr-bluetooth.
Is there any guide how to write sniffer using gr-bluetooth and libBTBB?

I saw some documentation about gr-bluetooth in the Wiki:
https://github.com/greatscottgadgets/ubertooth/wiki/Bluetooth-Captures-in-PCAP#BREDR

> Since PCAP has general applicability to packet capture, it made sense to provide a vendor-neutral, generic view of Bluetooth capture. The allocated DLTs can be applied to current and future versions of both ubertooth and gr-bluetooth, and potentially other RF capture tools as well.

So i assume that might be answer to my question.
Thank so much!
Tyler.

nic-kaczinsky, 2020-07-14:
@tylerone1  how are you doing?
I'm currently working on the similar thing and I'm interested in your progress. Can we talk about it?

mikeryan, 2020-12-01:
https://github.com/greatscottgadgets/gr-bluetooth is the project you seek

closed 2020-12-01 by mikeryan