greatscottgadgets/ubertooth #400: libubertooth 1.06 required * solution *

user4584, 2020-03-15:
### Steps to reproduce
1. follow how to install ubertooth wiki for Release 2018-12-R1
2. after all install are complete running ubertooth-rx as a test will produce libubertooth 1.06 required
current API is libubertooth 1.05 

* Solution I found *
1) download source repo from: git clone https://github.com/greatscottgadgets/ubertooth
2) download current libbtbb: wget https://github.com/greatscottgadgets/libbtbb/archive/2018-12-R1.tar.gz -O libbtbb-2018-12-R1.tar.gz
3) run all install instructions as normal from build wiki 
4) for firmware update use instructions from firmware update page:
# install build tools
sudo apt-get install gcc-arm-none-eabi libnewlib-arm-none-eabi
# make firmware 
cd firmware/bluetooth_rxtx/
make
# flash firmware, I needed sudo, might just be my OS. 
sudo ubertooth-dfu -d bluetooth_rxtx.dfu -r
# run test command 
ubertooth-rx
# you now have libubertooth 1.06. 
# I hope this helps any other users running into similar version issue. 

### Expected behaviour
ubertooth-rx should work, shouldn't be any issues.

### Actual behaviour
libubertooth is on version 1.05, but 1.06 is required 

### Version information
Debian 9 VM, specifically Parrot OS 

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (git-f2c5217), libbtbb 1.0 (2018-06-R1)

**libbtbb version:**
1.1
**Ubertooth firmware version (ubertooth-util -v):**
API 1.05
If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Ubertooth API version 1.05 found, libubertooth 1.1 requires 1.06.
Please upgrade to latest released firmware.
See: https://github.com/greatscottgadgets/ubertooth/wiki/Firmware

user4584, 2020-03-15:
I figured out a solution to my problem, just wanted to open the issue with my solution in case other users ran into the same problem. I hope this helps other users.

closed 2020-03-15 by user4584

chuzairps, 2022-06-29:
> ubertooth-rx
hi there. I am facing the same issue. How can I get rid of this. Any solution???Please help.
# ubertooth-rx                                                                                                                                                   127 ⨯
Ubertooth API version 1.00 found, libubertooth 1.1 requires 1.07.
Please upgrade to latest released firmware.
See: https://github.com/greatscottgadgets/ubertooth/wiki/Firmware