greatscottgadgets/ubertooth #401: ubertooth-rx only capture 22 byte packets

toruishihara, 2020-03-16:
### Steps to reproduce
1. Install and start ubertooth-rx -qa.pcap on 
2. Connect iPhone8 and SONY WH-1000XM3, or other Bluetooth BR/EDR connection
3. a.pcap only has 22 byte BR packets. No connection packets

### Expected behaviour
3. a.pcap include start connection and other packets.

### Actual behaviour
3. a.pcap only has 22 byte BR packets. No connection packets

### Version information
Ubuntu16.04TLS

**Ubertooth tools version (ubertooth-rx -V):**
Ubertooth 1.1 (2018-12-R1)

**libbtbb version:**
libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**
Ubertooth 2018-12-R1 (mikeryan@steel) Dec 4 2018

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Wireshark 2.6.10

### Output
```
Insert any commandline or build output here
```

tommcq, 2020-04-14:
need to specify uap to ubertooth-rx to follow a connection

mikeryan, 2020-12-01:
#427 might address this

closed 2020-12-01 by mikeryan