greatscottgadgets/ubertooth #404: Packet lost in LE following mode due to connection being limited to 2 packets

ffontaine, 2020-04-19:
### Steps to reproduce
1. Start ubertooth-btle -t AA:BB:CC:DD:EE:FF -f -c mypcap with any "modern" BLE devices

### Expected behaviour
All packets should be captured

### Actual behaviour
**Some packets will be missed** (for example `LL_FEATURE_REQ` sent in the same connection event than `CONNECT_REQ` will be missed). The issue is linked to the LE following mechanism which has been added in commit 6bfda52c0a769e04da14ba8be297eb7f429b6510. This mechanism assumes that a **connection event contains at most 2 packets**. This is not true anymore with any modern BLE devices (iOS, Android, RPi). Indeed these devices can send 4 or even 6 packets by connection event: https://punchthrough.com/maximizing-ble-throughput-on-ios-and-android. This is especially true when MTU is set to a high value (256, 512 ...). I don't know how to fix it properly. One option would be to check if `More Data` field is set to 1 in each packet before hopping to next channel. An other option would be to just wait on the channel until the end of the connection interval as the old LE following code is bluetooth_rxtx.c (prior to commit 218a24d35da3a7b32024482a539a33aa00996686) is not affected by this issue.

### Version information
**Operating system**: Linux

**Ubertooth tools version (ubertooth-rx -V):** 1.1 (git-f2c5217)

**libbtbb version:** 1.0 (git-6c7b9ff)

**Ubertooth firmware version (ubertooth-util -v):** git-f2c5217 (API:1.06)

mikeryan, 2020-12-01:
Do you have any example apps or devices that do this? Looking at MD is the right way to do it.

Are those high MTU packets across several L2CAP fragments?

ffontaine, 2020-12-01:
I found this issue with one of the D-Link camera (the DCS-8526LH), I updated ubertooth to manage the `More Data` field but unfortunately I wasn't able to make it work. Even with these changes, after some time with such high MTU packets (e.g.  when the DCS-8526LH sends back the list of WiFi networks to the app), ubertooth does not receive any packets. I assume that this a hardware issue that could be linked to the FIFO being full.

mikeryan, 2020-12-05:
$90 for that camera is a bit pricey for me, please let me know if you find any other devices that exhibit this behavior. I'm working with the BlueZ team now to see if there's a way to do this with fully open source tools so I can repro it easily enough to write a fix.

closed 2021-05-28 by straithe