greatscottgadgets/ubertooth #408: fails to compile with "-fno-common " or gcc-10

jonasstein, 2020-06-06:
### Steps to reproduce
1. compile either with CFLAGS="-march=native -O2 -pipe -fno-common" or use gcc-10

reported on gentoo: https://bugs.gentoo.org/706598
Confirmed today with the latest source from github.

```
[..]
/usr/bin/x86_64-pc-linux-gnu-gcc -fPIC -march=native -O2 -pipe -fno-common -Wl,-O1 -Wl,--as-needed -shared -Wl,-soname,libubertooth.so.1 -o libubertooth.so.1.1 CMakeFiles/ubertooth.dir/ubertooth.c.o CMakeFiles/ubertooth.dir/ubertooth_callback.c.o CMakeFiles/ubertooth.dir/ubertooth_control.c.o CMakeFiles/ubertooth.dir/ubertooth_fifo.c.o  -lusb-1.0 -lbtbb 
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: CMakeFiles/ubertooth.dir/ubertooth_callback.c.o:(.bss+0x0): multiple definition of `packet_counter_max'; CMakeFiles/ubertooth.dir/ubertooth.c.o:(.bss+0x10): first defined here
collect2: error: ld returned 1 exit status
[..]
```

jonasstein, 2020-06-06:
I guess the problem is here in the multiple definitions of `packet_counter_max`

```
grep -R packet_counter_max
libubertooth/src/ubertooth.c:unsigned int packet_counter_max;
libubertooth/src/ubertooth_callback.c:unsigned int packet_counter_max;
libubertooth/src/ubertooth_callback.c:		if((counter - last_seen[i] >= packet_counter_max)) {
libubertooth/src/ubertooth_callback.c:		if((counter - last_seen[i] >= packet_counter_max)) {
ubertooth-tools/src/ubertooth-afh.c:extern unsigned int packet_counter_max;
ubertooth-tools/src/ubertooth-afh.c:	packet_counter_max = 5;
ubertooth-tools/src/ubertooth-afh.c:			packet_counter_max = atoi(optarg);
```

ZeroChaos-, 2020-06-08:
I'm going to point out that this is a fairly significant fix, and a release hasn't been made in a while... maybe after this lands is a good time for a new tag?

ZeroChaos-, 2020-06-12:
Compile tested and basic runtime tested, this seems to work.

ZeroChaos-, 2020-06-12:
tested ubertooth-rx -z and ubertooth-btle -f, both of which scrolled the console in a reliable and error free way.

closed 2020-06-12 by mikeryan