greatscottgadgets/ubertooth #411: ubertooth-specan -d - -l 2400 -u 2483 -U -1 error

M3ASHKANANI, 2020-06-26:
### Steps to reproduce
1. start ubertooth-specan -d - -l 2400 -u 2483 -U -1
2. start ubertooth-rx

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
1.lower: 2402
upper: 2480
rx_xfer status: Transfer error. (1)

2. rx_xfer status: Transfer error. (1)

### Version information
RETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2020.2"
VERSION_ID="2020.2"
5.6.0-kali2-amd64

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:**
libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2018-12-R1 (API:1.06)

mikeryan, 2020-12-05:
Are you running Kali in a VM?

mikeryan, 2020-12-12:
Closing this due to inactivity. Please reopen if you're still having this problem.

closed 2020-12-12 by mikeryan