greatscottgadgets/ubertooth #412: DIY hardware is having problems

BG7YWL, 2020-07-06:
### Version information
**Operating system**: Raspberrypi 3B Linux 4.19.118-v7+

**Ubertooth tools version**: ubertooth-2018-12-R1

**libbtbb version**: libbtbb-2018-12-R1

### Steps to reproduce
I make the hardware with the Version 218 Gerber file and use the assembly_test firmware for my test.

It was found that only 1V8 LED(D1) and TX LED(D2) were lit. Then I used a multimeter to measure the 74 and 73 pins of LPC1754 chip, and found that the voltage was only 2.4mV, and the voltage of 73 and 72 pins was 3.1V.

[image]

In addition, I tested the USB_test program and found that the USB LED(D6) could not be lit. Could you please tell me how to troubleshoot the problem?

[image]

miek, 2020-07-15:
The best way to troubleshoot is to follow the guide for the assembly_test firmware here under 'Automated Test': https://github.com/greatscottgadgets/ubertooth/tree/master/firmware/assembly_test

It sounds like step 6 is not happening as described ("RX" (D3) is not illuminating)? Are the steps 2-5 passing correctly?

BG7YWL, 2020-07-16:
There was a problem in the second step, TX (D2), RX (D3), USR (D4) did not flash, check the adjacent resistance, the resistance value is normal, the pin connected to U4 chip is also normal, but the voltage is only 2.6mv

BG7YWL, 2020-07-17:
I measured the voltage of 6 pin of U1 and 51 pin of U4, and found that there is only 2.4mv, the normal should be 3.3V

closed 2020-07-22 by BG7YWL