greatscottgadgets/ubertooth #413: ubertooth-btle: lell_pcap_ppi_create_file: Permission denied

J053Fabi0, 2020-07-15:
### Steps to reproduce

    

1. Run the command: `mkfifo /tmp/pipe`
2. Open Wireshark
3. Click Capture -> Options
4. Click "Manage Interfaces" button on the right side of the window
5. Click the "New" button
6. In the "Pipe" text box, type "_/tmp/pipe_"
7. Click Save, then click Close
8. Click "Start"

In a terminal, run `ubertooth-btle`:
`sudo ubertooth-btle -f -c /tmp/pipe`

### Expected behaviour
In the Wireshark window you should see packets scrolling by.

### Actual behaviour
`ubertooth-btle: lell_pcap_ppi_create_file: : Permission denied`

### Version information
**Operating system**: Arch Linux

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 1.1 (git-de72ed9), libbtbb 1.0 (git-6c7b9ff)

**Ubertooth firmware version (ubertooth-util -v):** Firmware version: 2018-12-R1 (API:1.06)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

```
$ wireshark -v
Wireshark 3.2.5 (Git commit ed20ddea8138)

Copyright 1998-2020 Gerald Combs <gerald@wireshark.org> and contributors.
License GPLv2+: GNU GPL version 2 or later <https://www.gnu.org/licenses/gpl-2.0.html>
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiled (64-bit) with Qt 5.15.0, with libpcap, with POSIX capabilities (Linux),
with libnl 3, with GLib 2.64.4, with zlib 1.2.11, without SMI, with c-ares
1.16.1, with Lua 5.2.4, with GnuTLS 3.6.14 and PKCS #11 support, with Gcrypt
1.8.6, with MIT Kerberos, with MaxMind DB resolver, with nghttp2 1.41.0, with
brotli, with LZ4, with Zstandard, with Snappy, with libxml2 2.9.10, with
QtMultimedia, without automatic updates, with SpeexDSP (using system library),
with SBC, with SpanDSP, with bcg729.

Running on Linux 5.7.8-arch1-1, with Intel(R) Core(TM) i7-6600U CPU @ 2.60GHz
(with SSE4.2), with 15892 MB of physical memory, with locale es_ES.UTF-8, with
libpcap version 1.9.1 (with TPACKET_V3), with GnuTLS 3.6.14, with Gcrypt 1.8.6,
with brotli 1.0.7, with zlib 1.2.11, binary plugins supported (0 loaded).

Built using gcc 10.1.0.
```

### Output
```
$ sudo ubertooth-btle -f -c /tmp/pipe
ubertooth-btle: lell_pcap_ppi_create_file: : Permission denied
```

ZakB-K, 2020-08-31:
I believe this issue is related to group permission. For example, this issue was solved for me on Ubuntu by adding myself to the plugdev group, `sudo usermod -a zak -G plugdev`. This fix was inspired by issue #255. On Arch the `plugdev` group does not exist so I instead used the `uucp` group, `sudo usermod -a zak -G uucp`. I believe this will fix your issue too. 

The `plugdev` group defines who can plug devices in on Ubuntu and the `uucp` group defines similar permissions on Arch.

mikeryan, 2020-12-01:
Can you please provide the output of `ls -l /tmp/pipe`?

mikeryan, 2020-12-19:
Closing this due to inactivity. If you're still having this problem, please reopen the issue.

closed 2020-12-19 by mikeryan

iibanezgordon, 2021-08-12:
Hi, I am experiencing the same problem.

(The reproducing steps are the same)

sudo ubertooth-btle -f -tXX:XX:XX:XX:XX:XX -c /tmp/pipe

I got:
ubertooth-btle: lell_pcap_ppi_create_file: : Permission denied

I added the user to plugdev, but no results.

when I get ls -lh on /tmp/pipe
prwxr---w- 1 user user  0 Aug 12 16:41 /tmp/pipe

I did not had exec permissions on the pipe, I had added them (again without fixing the issue)

I am still digging into it, but any suggestion will be appreciatted

many thanks in advance